micropython/micropython-lib #379: urequests: Compatibility with CPython

amotl, 2020-03-25:
Hi there,

when invoking the `urequests` module on CPython, it will croak with
```
TypeError: %b requires a bytes-like object, or an object that implements __bytes__, not 'str'
```
when starting to write to the socket stream:
```
s.write(b"%s /%s HTTP/1.0\r\n" % (method, path))
```

This has also been reported on https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/10 and was fixed by https://github.com/adafruit/Adafruit_CircuitPython_Requests/issues/11.

With kind regards,
Andreas.

mattytrentini, 2022-07-03:
@andrewleech I think this is also still valid for urequests though I haven't tried it on CPython.

andrewleech, 2022-07-04:
urequests is currently not really compatible with CPython at all.
For a start, the import would need to handle usocket vs socket, then on cpython there is no `socket.write` function, only `socket.send`.  

Whether `urequests` should be compatible with cpython is an interesting discussion, which there could be conveniences to being able to test code in cpython, not much of the micropython-lib codebase would be compatible currently.

Separately to that, the micropython `b"%s" % str` clearly has an auto-convert which cpython does not support - I guess this is a confusing difference between the platforms, though I feel it's one of many cases micropython is more "tolerant" of things than cpython.

dpgeorge, 2022-10-04:
I don't think it's worth supporting `urequests` (or any other library in this repo) running under CPython.  That would be too restrictive and not allow us to implement things in an efficient way.

closed 2022-10-04 by dpgeorge