micropython/micropython-lib #380: upip mbedtls_ssl_handshake

bert2002, 2020-03-29:
Hi,
i am trying to use upip on a ESP32 and it is failing with
```
MicroPython v1.12-317-g688323307 on 2020-03-29; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import upip
>>> upip.install('pkg_resources')
Installing to: /lib/
mbedtls_ssl_handshake error: -4290
Error installing 'pkg_resources': [Errno 5] EIO, packages may be partially installed
>>> 
```
I tried firmwares `esp32-idf3-20200328-v1.12-310-g9418611c8.bin`, `esp32-idf3-20200329-v1.12-317-g688323307.bin` and `esp32-idf4-20200329-v1.12-317-g688323307.bin`, but getting the same errors.  The ESP32 is connected to the wifi. 

Any idea why this is happening?

Many thanks,
bert

george-hawkins, 2020-04-29:
Hello @bert2002 - I'm not involved with the MicroPython project but this repo is largely unmaintained (though this might change at some stage, see #377). So you're probably better off logging issues at [micropython/micropython](https://github.com/micropython/micropython/issues).

I and others have seen similar problems to you - the tracking issue for this seems to be micropython/micropython#5543. At the moment it doesn't look like this is a fairly long term issue with the ESP32 port, unfortunately.

bert2002, 2020-05-01:
Thanks for the heads up @george-hawkins ;)

closed 2020-05-01 by bert2002