micropython/micropython-lib #384: HMAC AttributeError

cositotito, 2020-04-24:
hello all,
I have a problem when using 'import hmac' and I try to use it on my esp32

* before using it, I install it and it installs correctly, the code is the following:
import upip
upip.install ("hmac")

* the code is the following to use hmac:
digest = hmac.new (s.decode ("base64"), a, hashlib.sha3_256) .digest ()

* the error that prints to me on my console:
Traceback (most recent call last):
   File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'new'

How can I use the new or remove this error?
I would appreciate the help

cositotito, 2020-04-24:
* Code example.py:

import hmac
import hashlib
import time
import urequests
import json

secret="345fg-dfg"
word="536"
signature = hmac.new(secret, word.encode(), digestmod=hashlib.sha256).hexdigest().upper()

* return Error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/hmac.py", line 149, in new
  File "/lib/hmac.py", line 63, in __init__
AttributeError: 'sha256' object has no attribute 'digest_size'

mvo5, 2021-07-21:
Fwiw, I ran into the same issue and pushed a PR that fixes this for me. Let's see what the code review says.

dpgeorge, 2022-08-08:
Should be fixed by f95260d7e3c123a8b1907ded095e4c25e4251085

closed 2022-08-08 by dpgeorge