micropython/micropython-lib #386: MQTT connection and client.check_msg() problem

rodrigo1120, 2020-05-12:
I'm using Adafruit IO switches to turn my nodeMCU lights ON and OFF. The following code works but for a limited period of time. 5 minutes after the connection into Adafruit IO it seems the client tries to reconnect and if this happens I lose my switch funcionality at my adafruit dashboard. Either if I try to turn on or off the LED on my dashboard, the nodeMCU does not responds accordingly. Here is the code used (I'm using micropython, pycharm and a Lolin v3 nodemcu)

```
from machine import Pin
import sys
import os
import network
from umqtt.robust import MQTTClient

random_num = int.from_bytes(os.urandom(3), 'little')
mqtt_client_id = bytes('client_' + str(random_num), 'utf-8')

ADAFRUIT_LINK = b'io.adafruit.com'
ADAFRUIT_USERNAME = b'rodrigo1120'
ADAFRUIT_KEY = b'aio_LecA91KhBP28IBxCV8fDogtfurHy'
ADAFRUIT_FEEDNAME1 = b'gpio16'
ADAFRUIT_FEEDNAME2 = b'gpio2'

ssid = "4G"
password = "4G@L1N32069"

client = MQTTClient(client_id= mqtt_client_id,
                    server= ADAFRUIT_LINK,
                    user=ADAFRUIT_USERNAME,
                    password=ADAFRUIT_KEY)

mqtt_feedname1 = bytes('{:s}/feeds/{:s}'.format(ADAFRUIT_USERNAME, ADAFRUIT_FEEDNAME1), 'utf-8')
mqtt_feedname2 = bytes('{:s}/feeds/{:s}'.format(ADAFRUIT_USERNAME, ADAFRUIT_FEEDNAME2), 'utf-8')

def connect_to_ap():
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    if not wlan.isconnected():
        print('Connecting to {}'.format(ssid))
        wlan.connect(ssid, password)
        while not wlan.isconnected():
            pass
    print('Credentials: ', wlan.ifconfig())
    try:
        client.connect()
    except Exception as e:
        print('Could not connect to MQTT Server {}{}'.format(type(e).__name__, e))

def callback_routine(feed, msg):
    print('Received Data from: feed={}, msg={}'.format(feed, msg))
    if ADAFRUIT_FEEDNAME1 in feed:
        action = str(msg, 'utf-8')
        if action == 'ON':
            pin16.value(0)
        else:
            pin16.value(1)
        print('Action: {}'.format(action))
    if ADAFRUIT_FEEDNAME2 in feed:
        action = str(msg, 'utf-8')
        if action == 'ON':
            pin2.value(0)
        else:
            pin2.value(1)
        print('Action: {}'.format(action))

pin16 = Pin(16, Pin.OUT)
pin2 = Pin(2, Pin.OUT)
pin16.value(1)
pin2.value(1)

def main():
    connect_to_ap()
    while True:
        client.set_callback(callback_routine)
        client.subscribe(mqtt_feedname1)
        client.subscribe(mqtt_feedname2)
        try:
            client.wait_msg()
        except KeyboardInterrupt:
            print('Ctrl+C pressed, exiting')
            client.disconnect()
            sys.exit()

if __name__ == '__main__':
    main()
```

Here is a picture of my serial monitor with this problem: (I took this picture 15 minutes after I stopped using the adafruit IO switches)
[image]

EddieParis, 2020-05-15:
Try to pull the change referenced here in PR #385 I had same issue, this is the solution.

rodrigo1120, 2020-05-15:
> Try to pull the change referenced here in PR #385 I had same issue, this is the solution.

In order for this to work, I just need to upload this robust.py inside my nodemcu and use `import robust.py`?

EddieParis, 2020-05-15:
Yes, Ugly but easy way to do that is to copy/paste the file from my fork, you can also clone my fork or get the patch and apply it on you clone.

rodrigo1120, 2020-05-16:
> Yes, Ugly but easy way to do that is to copy/paste the file from my fork, you can also clone my fork or get the patch and apply it on you clone.

I tried to copy/paste your code, but unfortunately I got this error:
[image]

in pycharm they say I can install package simple, but if I do this I got an install package error.
I'm sorry, I guess I'm struggling with a simple relative import problem

rodrigo1120, 2020-05-16:
I did the following changes in the code, but I got no fix for that initial problem:

```
import utime
from umqtt.robust import simple
# micropython-lib/umqtt.robust/umqtt/robust.py

class MQTTClient(simple.MQTTClient):

    DELAY = 2
    DEBUG = False

    def delay(self, i):
...
```
(Inside robust.py)

```
from machine import Pin
import sys
import os
import network
# from umqtt.robust import MQTTClient
from robust import MQTTClient

random_num = int.from_bytes(os.urandom(3), 'little')
mqtt_client_id = bytes('client_' + str(random_num), 'utf-8')

ADAFRUIT_LINK = b'io.adafruit.com'
ADAFRUIT_USERNAME = b'rodrigo1120'
ADAFRUIT_KEY = b'AAAAAAAAAAAAAAAAAAAAAAAAAA'
ADAFRUIT_FEEDNAME1 = b'gpio16'
ADAFRUIT_FEEDNAME2 = b'gpio2'

ssid = "WIFI-USER"
password = "WIFI-PASSW"

client = MQTTClient(client_id= mqtt_client_id,
                    server= ADAFRUIT_LINK,
                    user=ADAFRUIT_USERNAME,
                    password=ADAFRUIT_KEY,
                    ssl=False)
...
```
(Inside main.py)

EddieParis, 2020-05-17:
Sorry, I was mislead by the title of your PR. You don't use check_msg(), but wait_msg(). My patch is specifically for check_msg() that can get stuck, so you may have a different issue.

It might be the well known loosing subscriptions issue. If the robust layer reconnects, you loose all subscriptions. Try connecting with a non clean session so that you broker keep track of them.

I recommend reading this chapter :
https://www.hivemq.com/blog/mqtt-essentials-part-7-persistent-session-queuing-messages/

And the whole essentials articles from this website are worthy too.

jonnor, 2024-08-25:
Is this still an issue on the latest MicroPython and mqtt.simple version?