micropython/micropython-lib #387: Umqtt.simple re-subscribe issue

samf48, 2020-05-14:
Hello I have a device in which I am using this library to subscribe and publish.  The publish works well an the subscribe works IF my device is actively on and connected.  What I mean by this is that if I am subscribed to a topic and I publish to said topic, then all is well.  However my device actually powers down in a cyclic fashion and only powers up when it wants to publish.  So with that said, how can I get messages from my subscribed topic once my device powers back up and then "re-subscribes" ?  I'm pretty sure someone is doing this already but I haven't found a clear way to do it. Or is this something that is dependent on the server side?

HerbV, 2020-06-08:
hi samf48, I wold just publish a message to be informed about the new start. What do you think?

jonnor, 2024-08-25:
To get the last message on a subscribe, the "retained message" feature in MQTT must be used. See for example https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/ for an explainer. The client that publishes the message must set the retain flag.

closed 2024-08-25 by robert-hh