micropython/micropython-lib #395: AttributeError: 'NoneType' object has no attribute 'func'

nick-0307, 2020-06-27:
When I import multiprocessing, there is an error saying that:   
  File "main.py", line 5, in <module>
  File "multiprocessing.py", line 3, in <module>
  File "select.py", line 14, in <module>
 AttributeError: 'NoneType' object has no attribute 'func'

Would you please help me solve this problem?

kitplummer, 2022-09-06:
I get the same with the `base64` module:

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "/lib/base64.py", line 9, in <module>
  File "/lib/re.py", line 11, in <module>
AttributeError: 'NoneType' object has no attribute 'func'
```

andrewleech, 2022-09-06:
In both cases here the ffi modules for select and re respectively have been installed, but they can't find the referenced dynamic libraries.

@kitplummer if you see `re.py", line 11,` in https://github.com/micropython/micropython-lib/blob/f3cfc52ab076fc913dc6e266fb7370b418c8f542/unix-ffi/re/re.py#L11 it's clear `pcre` is None meaning the ffi command earlier couldn't find it https://github.com/micropython/micropython-lib/blob/f3cfc52ab076fc913dc6e266fb7370b418c8f542/unix-ffi/re/re.py#L6

andrewleech, 2022-09-06:
@jimmo would it make most sense for `ffilib.open()` raise an exception if it can't find the library?

kitplummer, 2022-09-06:
@andrewleech - thanks, eventually dug in enough to find that being the issue.

I'm working with a Raspberry Pico W, which apparently doesn't have the lib.  :)  Working around it now, but that might be worth adding to make it obviously clear.

andrewleech, 2022-09-06:
Ah yes, the ffi based libs only work on Linux.

How did you install these libraries out of interest?

This repo is in the process of being reorganised a bit with the recent manifest file charges, working towards making library compatibility and installation easier and more reliable.

kitplummer, 2022-09-06:
I installed with `upip`.

kitplummer, 2022-09-06:
For posterity, I was trying to use urllib.parse for `urlencode` purposes - which pulls in `micropython-re-pcre` (and the ffilib dep) by default.

andrewleech, 2022-09-06:
Ah thanks, that context definitely helps! `urllib.parse` is definitely something I can see as useful on rpi-w / networked boards, it'd be worth updating it to work with the built-in `re` module.

kitplummer, 2022-09-07:
Should've mentioned, the quick-fix for me was to simply delete the re.py and ffilib.py files from the Pico's lib/ subdir - and using the built-in stuff.

jonnor, 2024-08-25:
There seems to have been a fix/clarification provided. Proposing to close this issue.

closed 2024-08-25 by mattytrentini