micropython/micropython-lib #405: Example for uasyncio.websocket.server fails

damiencorpataux, 2020-12-07:
I need to run a websocket server on ESP32 and [the official example](https://github.com/micropython/micropython-lib/blob/master/uasyncio.websocket.server/example_websock.py) raises the following exception when I connect from any client:
```
MPY: soft reboot
Network config: ('192.168.0.200', '255.255.255.0', '192.168.0.1', '8.8.8.8')
b'Sec-WebSocket-Version: 13\r\n'
b'Sec-WebSocket-Key: k5Lr79cZgBQg7irI247FMw==\r\n'
b'Connection: Upgrade\r\n'
b'Upgrade: websocket\r\n'
b'Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits\r\n'
b'Host: 192.168.0.200\r\n'
b'\r\n'
Finished webrepl handshake
Task exception wasn't retrieved
future: <Task> coro= <generator object 'echo' at 3ffe79b0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "main.py", line 22, in echo
  File "uasyncio/websocket/server.py", line 60, in WSReader
AttributeError: 'Stream' object has no attribute 'ios'
```

My micropython firmware and libraries:
* Micropython firmware: https://micropython.org/resources/firmware/esp32-idf3-20200902-v1.13.bin
* Pip libraries installed: micropython-ulogging, uasyncio.websocket.server

My `main.py`:
```
import network
import machine

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.ifconfig(('192.168.0.200', '255.255.255.0', '192.168.0.1', '8.8.8.8'))
if not sta_if.isconnected():
    print('connecting to network...')
    sta_if.connect('my-ssid', 'my-password')
    while not sta_if.isconnected():
        machine.idle() # save power while waiting
print('Network config:', sta_if.ifconfig())

# from https://github.com/micropython/micropython-lib/blob/master/uasyncio.websocket.server/example_websock.py
import uasyncio
from uasyncio.websocket.server import WSReader, WSWriter

def echo(reader, writer):
    # Consume GET line
    yield from reader.readline()

    reader = yield from WSReader(reader, writer)
    writer = WSWriter(reader, writer)

    while 1:
        l = yield from reader.read(256)
        print(l)
        if l == b"\r":
            await writer.awrite(b"\r\n")
        else:
            await writer.awrite(l)

import ulogging as logging
#logging.basicConfig(level=logging.INFO)
logging.basicConfig(level=logging.DEBUG)
loop = uasyncio.get_event_loop()
loop.create_task(uasyncio.start_server(echo, "0.0.0.0", 80))
loop.run_forever()
loop.close()
```

beyonlo, 2022-01-14:
Hello @damiencorpataux 

Did you success run any example of uasyncio websocket server? I need to run a websocket server on ESP32 too, but I would like in secure mode, with SSL. But that official link in your example https://github.com/micropython/micropython-lib/blob/master/uasyncio.websocket.server/example_websock.py do not exists anymore. And, in new MicroPython structured I not found too :(

Thank you.

QGB, 2022-10-06:
"Stream object has no attribute ios"

jimmo, 2022-10-06:
For async websocket server, see https://github.com/miguelgrinberg/microdot

closed 2022-10-06 by jimmo

QGB, 2022-10-06:
@jimmo how to use upip to install [microdot](https://github.com/miguelgrinberg/microdot) ?

```
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing microdot 1.2.0 from https://files.pythonhosted.org/packages/d5/69/80d1e1b3dfab405de13e6e3dff42e286cf5f03d28e85b2a15341ccecbd57/microdot-1.2.0.tar.gz
>>> import microdot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "microdot.py", line 25
SyntaxError: invalid syntax
```

jimmo, 2022-10-06:
> @jimmo how to use upip to install [microdot](https://github.com/miguelgrinberg/microdot) ?

@QGB  upip has been essentially unsupported for a few years. I don't believe microdot has ever been installable by `upip`.

Coincidentally, I just published `upip`'s replacement, `mip`, a few days ago. See 
https://github.com/miguelgrinberg/microdot/issues/67 for discussion about making it installable via `mip`.

In the meantime you can install manually by copying the required files (either just microdot.py, or both microdot.py and microdot_asyncio.py if you want the async version).
See
https://microdot.readthedocs.io/en/latest/intro.html#running-with-micropython
https://microdot.readthedocs.io/en/latest/extensions.html#asynchronous-support-with-asyncio