micropython/micropython-lib #413: Unable to open SSL / HTTPS socket with ESP8266: OSError from ssl.wrap_socket

joseph-long, 2021-02-03:
I've just set up an ESP8266 with 4MB of flash using `esp8266-20210202-v1.14.bin` and I'm trying to run some example code to open a socket for https:

```
import usocket
import ssl

hostname = 'xwcl.science'
s = usocket.socket()
ai = usocket.getaddrinfo(hostname, 443)
addr = ai[0][-1]
print(addr)
s.connect(addr)
s = ssl.wrap_socket(s)
```

In the REPL, it all goes well until `ssl.wrap_socket`:

```
>>> hostname = 'xwcl.science'
>>> s = usocket.socket()
>>> ai = usocket.getaddrinfo(hostname, 443)
>>> addr = ai[0][-1]
>>> print(addr)
('104.238.141.27', 443)
>>> s.connect(addr)
>>> s = ssl.wrap_socket(s)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -40
```

jonnor, 2024-08-25:
Hi. There has been many improvements to ESP8266 and ssl over the last years. Is this still a problem on latest MicroPython version?

jonnor, 2024-08-25:
There is more information, including tests on recent MicroPython in https://github.com/micropython/micropython-lib/issues/400
Proposing to close this issue in favor of that one.

closed 2024-08-25 by mattytrentini