micropython/micropython-lib #414: unittest.main fails to discover tests

OpenBagTwo, 2021-02-17:
Running:
```python
>>> import unittest
>>> unittest.main('my_package.tests.test_module')
```
results in
```
Ran 0 tests

OK
```

The workaround is to install `test.support` and utilize `run_unittest` but wow that's a lot of overhead.

The fix also appears to be to use [the four-argument version of `__import__`](https://docs.python.org/3/library/functions.html#__import__):

Replace:
https://github.com/micropython/micropython-lib/blob/eae01bd4e4cd1b22d9ccfedbd6bf9d879f64d9bd/unittest/unittest.py#L217

with:

```python
    m = __import__(module, None, ['object'], True)
```

Issue present with:
- MicroPython
  - v1.14 on 2021-02-02; ESP module with ESP8266
  - MicroPython 3d45bca-dirty on 2021-02-02; linux version
- micropython-unittest 0.3.2

OpenBagTwo, 2021-04-11:
Ah. Just discovered that this version of `__import__` causes `MemoryError`s on an ESP82xx boards (encountered using MicroPython v1.14 on 2021-02-02; ESP module (1M) with ESP8266). I'll go ahead and close this and figure out a better way to do unit test discovery.

OpenBagTwo, 2021-04-11:
Never mind. The issue was actually that my unit tests themselves exhausted the microcontroller's memory--I can't even directly import the test module. So the four-arg version of `__import__` itself is not the problem.

mattytrentini, 2022-07-03:
There has been some significant improvements in the unittest library recently - including around discovery. See #488 for details. Could you please update to the latest unittest library and try again?

dpgeorge, 2022-10-04:
This should be fixed now that #488 is merged.

closed 2022-10-04 by dpgeorge