micropython/micropython-lib #420: Replace triple quotes with actual comments

ThomasTheGerman, 2021-04-28:
In many places in these libraries, there are huge "block comments" in the form of triple quotes, which cause memory issues, as Micropython tries to allocate them as a continuous block on the heap. These should be replaced with the proper python comment style.

stinos, 2021-04-28:
Fair point, but there's a difference between a 'proper' Python comment and a [docstring](https://www.python.org/dev/peps/pep-0257/) which is what the triple quotes do. So when you have a piece of code with docstrings and use it with CPython, those comments are available via the `__doc__` special method for instance, can be used to generate documentation automatically etc. Which also matters because there are enough cases where it is interesting to have Python code which can be used by both CPython and MicroPython. But since (as far as I know) MicroPython doesn't actually use dosctrings a better solution would perhaps be to change the parser to not allocate them. Or at least have an option for that.

stinos, 2021-05-11:
Also see https://github.com/micropython/micropython/pull/3748 which comes to the same conclusion: this should be fixed in the core.

jonnor, 2024-08-25:
Is this still a problem with latest MicroPython (currently 1.23)?