micropython/micropython-lib #429: OSError 16 when executing urequests.post()

netomarin, 2021-06-10:
I'm trying to send data to a spreadsheet hosted on Google using the post function on urequests.py

my code:
```
request_data = ujson.dumps({
    "parameter": {
        "Beer": "Test",
        "Temp": 65,
        "SG":1.050,
        "Color":"ORANGE",
        "Comment":"user@gmail.com, TiltESP32",
        "Timepoint":43486.6
    }
})

print("url: {} - data: {}".format(device['cloud_url'], request_data))
response = urequests.post(device['cloud_url'], headers = {'content-type': 'application/json'}, data = request_data)
```
result:
```
url: https://script.google.com/macros/s/AKfycbwNXh6rEWoULd0vxWxDylG_PJwQwe0dn5hdtSkuC4k3D9AXBSA/exec - data: {"parameter": {"Comment": "user@gmail.com, TiltESP32", "Beer": "Test", "Temp": 65, "SG": 1.05, "Color": "ORANGE", "Timepoint": 43486.6}}
Traceback (most recent call last):
  File "main.py", line 108, in bt_irq
  File "main.py", line 46, in handle_ibeacon
  File "main.py", line 100, in log_data
  File "urequests.py", line 120, in post
  File "urequests.py", line 62, in request
OSError: 16
```
Here is the line where the error is hapening: https://github.com/micropython/micropython-lib/blob/2e91b924135deee7e287fb20bdd6b158a2914b66/python-ecosys/urequests/urequests.py#L62

I'm using a standard ESP-WROOM-32 board with the imagem file `https://micropython.org/resources/firmware/esp32-20210610-unstable-v1.15-202-g71e3538a3.bin`

jimmo, 2022-07-23:
OSError(16) is "device or resource busy"

Although most of the issues we see with wrap_socket are https://github.com/micropython/micropython/issues/8940 (we're working on a fix for this).

coldreason, 2022-07-24:
When i remove server_hostname parameter on wrap_socket at urequests
I found that error cleared.
But i am not sure side effects of it