micropython/micropython-lib #438: 'pip install micropython-machine' failed

DesperateForHelpIntern, 2021-08-10:
I ran the pip install command given by the machine main website and this error occured:  

ERROR: File "setup.py" not found for legacy project micropython-machine from https://files.pythonhosted.org/packages/e1/31/a3282942b3e96881393db406130735913c647f0fc9e07a0d60d7b546b06b/micropython-machine-0.2.tar.gz#sha256=b8747926ee044547c5f5df09b8015977b4137c744800285ee92f307acc477223.

Interestingly, a very similar error occured when I tried installing upip to install machine. (pip install micropython-upip) returned the following error: 

ERROR: File "setup.py" not found for legacy project micropython-upip from https://files.pythonhosted.org/packages/29/81/52df27cb7d240190c1d3ddab0c081bd3296bedcce155eed9e7016c4d92f0/micropython-upip-1.2.6.tar.gz#sha256=b9c09fc2b65f877d945142e468e0f25b5f70e9f74e98e0f43ca030bd7f47e87f.

DesperateForHelpIntern, 2021-08-11:
I tried installing machine again via thonny's package manger and this happened: C:\Users\sawye\AppData\Local\Programs\Thonny\python.exe -m pip install --no-cache-dir --user micropython-machine --disable-pip-version-check

Collecting micropython-machine
  Downloading micropython-machine-0.2.tar.gz (1.7 kB)
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\sawye\AppData\Local\Programs\Thonny\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\sawye\\AppData\\Local\\Temp\\pip-install-35eahy63\\micropython-machine\\setup.py'"'"'; __file__='"'"'C:\\Users\\sawye\\AppData\\Local\\Temp\\pip-install-35eahy63\\micropython-machine\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\sawye\AppData\Local\Temp\pip-pip-egg-info-4drkjt7o'
         cwd: C:\Users\sawye\AppData\Local\Temp\pip-install-35eahy63\micropython-machine\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\sawye\AppData\Local\Programs\Thonny\lib\tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\sawye\\AppData\\Local\\Temp\\pip-install-35eahy63\\micropython-machine\\setup.py'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Process returned with code 1

hellozepp, 2022-01-06:
Same error

tarekfoau, 2022-06-15:
same issue here
maybe i am installing it in a wrong way or it is just does need installation?

andrewleech, 2022-06-15:
I'm not sure how thonny is expected to work but currently these packages are not expected to work with `pip` because that's for cpython, whereas these packages can only be installed in micropython.

The official method I know of currently is to have a micropython build with `upip` frozen directly into it, then you can `micropython -m upip install micropython-machine`

tarekfoau, 2022-06-15:
> I'm not sure how thonny is expected to work but currently these packages are not expected to work with `pip` because that's for cpython, whereas these packages can only be installed in micropython.
> 
> The official method I know of currently is to have a micropython built with `upip` frozen directly into it (or `upip.py` manually added to current folder / micropython path) then you can `micropython -m upip install micropython-machine`
> 
> Alternatively this tool might help, though I haven't tried it myself: https://pypi.org/project/pipkin/

thank you very much <3

yatharthb97, 2022-07-18:
@tarekfoau Did that solve the issue for you?

andrewleech, 2022-07-18:
On a related note, the packaging system for micropython is in the first stages of being overhauled to hopefully resolve issues / confusion like this. For more details you might like to follow https://github.com/micropython/micropython/pull/8914

yatharthb97, 2022-07-18:
`pipkin -m upip install micropython-machine` broke my env.

Error traceback:

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/lib/machine/__init__.py", line 2, in <module>
>   File "/lib/machine/timer.py", line 7, in <module>
>   File "/lib/signal.py", line 13, in <module>
> AttributeError: 'NoneType' object has no attribute 'func'

Does anyone know how to fix this? Reinstalling thonny didn't help.

andrewleech, 2022-07-18:
Yeah ok, pipkin is a third party project that's attempting to make the most out of what's currently on pypi, most of which is old / unmaintained / full of inconsistencies.

To be honest the mechanism for publishing micropython packages to pypi has been broken and unused for a long time, hence the PR I linked earlier about building a new system that doesn't rely on pypi at all and is intended to be far more suitable for micropython users.

Without knowing what/how you've currently installed micropython I can't help much with cleaning it. Any other questions relating to pipkin and what it's installed would be better asked on the pipkin project pages.

andrewleech, 2022-07-20:
Thonny is also a third party project, I believe it's by the same author as pipkin. I don't use it sorry, you'll need to look up docs/forums/etc for thonny.

As I mentioned earlier there is work being done to fix/improve the official packaging install system (based on manifest files) but I don't know how long it's expected to take.

dpgeorge, 2022-10-04:
`upip` has been replaced with `mip`, and the way to install things on the unix port is being reworked to use `mip` as well.

Otherwise it looks like this is an issue related to thonny/pipkin, so best to raise that issue in the appropriate repository.

closed 2022-10-04 by dpgeorge