micropython/micropython-lib #453: AIOBLE notified() return None value when the peripheral is sending notificaiton values

steve1aa, 2021-10-01:
**PROBLEM:**
I have a peripheral that sending the notification value constantly, and no None value. But when I invoke the characterisitc.notified() method, I got an None value.
**BACKGROUND:**
The peripheral will send the notify event per 1 second.
During testing, the connection keeps alive.
**TEST CODE:**
`
data = await characteristic.notified()
print(data)
await asyncio.sleep_ms(2000)
data = await characteristic.notified()
print(data)
data = await characteristic.notified()
print(data)
data = await characteristic.notified()
print(data)
`
**RESULT:**
`
b'\n\x00'
b'\x14\x00'
None
b'\xb\x00'
`
**ANALYSIS:**
WHY here is a None return from notified()?
The key part is on "await asyncio.sleep_ms(2000) "after first notified().
I checked the code in client.py

1. When sleeping, notify events are still coming, so line 357 is still setting value to characteristc._notify_data, and line 360 is still set the notify event.
2. When second notified() be invoked, in line 343,  self._notify_data is not None(the data is front previous notify event), so it return immediately.
3. When third notified() be invoked, line 343, self._notify_data is None (it is consumed by step 2), then it jumps to the notify_event, but because the last notify_event is set in step 1, so in line 347, it return immediately without waiting for the REAL notify event, and the return data = None.
[image]

jimmo, 2021-11-08:
Thanks for the report @steve1aa -- please see https://github.com/micropython/micropython-lib/pull/459 for a fix.

dpgeorge, 2021-11-11:
Fixed by dc03b4af4d9b7e65f4d3a4f43691179ffb0bb39d

closed 2021-11-11 by dpgeorge