micropython/micropython-lib #466: umqtt.simple not working: MQTTException: 2

jberes-ae, 2021-11-19:
I am trying to run a demo program for Digi Xbee cellular modem to publish and subscribe MQTT packets.
* Error: File "/flash/lib/umqtt/simple.py", line 110, in connect
* running mosquitto 2.0.13 on Windows 10
* port 1883, so not authentication needed

Any insights to resolving would be greatly appreciated. Thank you.

[image]

[image]

chouex, 2021-12-20:
same message MQTTException: 2 

log from MQTT server eclipse-mosquitto(457fe4154880)
```
1639972746: New client connected from xxxx as mqtt-explorer-8c09a901 (p2, c1, k60).
1639972796: New connection from xxx on port 1883.
1639972796: New client connected from xxxx as ESP8266Client1 (p2, c1, k0).
1639972796: Bad socket read/write on client ESP8266Client1: Invalid arguments provided.

chouex, 2021-12-20:
[image]

by compare another client, seems new version can not just set keep alive to `0`
so change msg = bytearray(b"\x04MQTT\x04\x02\0\0") to msg = bytearray(b"\x04MQTT\x04\x02\0`\x3c`") #60s keep alive

tested 2.0.12 and latest

ref
https://github.com/micropython/micropython-lib/blob/3a6ab0b46d6471bee00ae815444c624709dd4cdd/micropython/umqtt.simple/umqtt/simple.py#L74

amotl, 2021-12-20:
Hi there,

the issue you might be observing here has been reported and discussed at #445 and https://github.com/hiveeyes/terkin-datalogger/pull/97 already. In this context, please be aware of [CVE-2020-13849](https://nvd.nist.gov/vuln/detail/CVE-2020-13849), which is the main reason for this change in Mosquitto 2.0.12:

```text
- Fix `max_keepalive` not applying to MQTT v3.1.1 and v3.1 connections.
  These clients are now rejected if their keepalive value exceeds
  max_keepalive. This option allows CVE-2020-13849, which is for the MQTT
  v3.1.1 protocol itself rather than an implementation, to be addressed.
```

Now, Mosquitto 2.0.13 adds the possibility for `max_keepalive` to be set to `0` again with https://github.com/eclipse/mosquitto/commit/d942ed7ee, if you really need it.

With kind regards,
Andreas.

amotl, 2021-12-20:
Dear @chouex,

> so change msg = bytearray(b"\x04MQTT\x04\x02\0\0") to msg = bytearray(b"\x04MQTT\x04\x02\0\x3c") #60s keep alive

There is no need to manipulate this on a binary/bytes level. There is already the `keepalive` option as a parameter to `MQTTClient`'s constructor. You can easily pass a custom value from your application code, or adjust the default value of the library like https://github.com/daq-tools/pycopy-lib/commit/fe68f4186.

With kind regards,
Andreas.

chouex, 2021-12-20:
Dear @amotl 

world you consider set keepalive value deafault to non-zero? 
I beleive many people like me just want to try the example and run into issue immediately.

oooeeee, 2022-02-17:
As I see on screenshot from @jberes-ae - your mosquito running in local mode, it listen only on localhost. You need to add `listener 1883 0.0.0.0` into your mosquitto.conf

jonnor, 2024-08-25:
This appears to be a duplicate of https://github.com/micropython/micropython-lib/issues/445 - proposing to close this issue.

closed 2024-08-25 by mattytrentini