micropython/micropython-lib #473: Get request fails with OSerror: -40

kaninfod, 2021-12-26:
I do the following request on a D1 mini esp8266:

```
import urequests
response = urequests.get('http://192.168.0.121:3033/api/status')

```

The server is running on my network as a node.js server.
It works as expected

When I expose the server to the internet through a Caddy redirect I end up with a new url which has HTTPS 

```
import urequests
response = urequests.get('https://<mydomain-here-but-sensored>/api/status')

```

This does not work:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 116, in get
  File "urequests.py", line 62, in request
OSError: -40
```

I see that File "urequests.py", line 62 is:

```
if proto == "https:":
            s = ussl.wrap_socket(s, server_hostname=host)
        s.write(b"%s /%s HTTP/1.0\r\n" % (method, path))
```

related to https - but I don't know what to do from here...
Any advice or workaround?

mattytrentini, 2022-07-03:
Duplicate of #400 (but thanks for the additional info, which is now linked)

closed 2022-07-03 by mattytrentini

Erkman84, 2022-11-08:
This issue still persists on Micropython 1.19.1
(I use POST) but it also does not work with get, it is combined with 'https' (SSL) domains.
  File "urequests.py", line 120, in post
  File "urequests.py", line 62, in request
OSError: -40

vencax, 2023-07-06:
Still persists on MP 1.20