micropython/micropython-lib #475: AIOBLE - ScanResult only returns first service when multiple are present in advertisement

rale, 2021-12-31:
When a device advertises multiple service UUIDs, it seems only the first one gets decoded. 

I'm doing something like:
```
async with aioble.scan(5000, interval_us=30000, window_us=30000, active=True) as scanner:
    async for result in scanner:
        for service in result.services():
            print(service)
```
With an advertising packet like this:
`020106050318182618`
I think result.services() should yield both 0x1818 and 0x1826, but it only seems to decode the first one.

jimmo, 2022-08-05:
> With an advertising packet like this:
> 020106050318182618
> I think result.services() should yield both 0x1818 and 0x1826, but it only seems to decode the first one.

Thanks, this is a bug, we do not current handle the decoding case where multiple service UUIDs are encoded into the same field (as an optimisation we should also handle this on the encode side too).

aetelani, 2024-05-11:
I don't get service data work wither, for example sending temperature values to central. Manufacturer information seems to work though.