micropython/micropython-lib #478: Copyleft violation

LarsKumbier, 2022-01-24:
The following file has been copied from this repository without any license information and violates the MIT license: 

https://github.com/RuiSantosdotme/ESP-MicroPython/blob/master/code/MQTT/umqttsimple.py

perkinslr, 2022-01-28:
Assume an honest mistake.  The project is on github, so open a PR to correct the missing license header.  Before you do:

Be sure it is really the same file, not just the same name and a similar API.  
Be sure it was MIT licensed two years ago, and not potentially sourced from somewhere under a different license.

Also note that the file in this repository similarly lacks copyright and license information, and the current (new) install instructions recommend copying individual files from this repository into other projects, rather than relying on the `Makefile` based install.  Since the files individually do not contain their license information, and the install instructions do not mention license implications of manually copying files, these problems are likely to be pretty common, especially with hobbiest projects.  

Finally, note the project in question hasn't been touched in 2 years, the particular file in question in 4 years.  I would suggest being extra polite, as it would not do *this* project any favors to make a spectacle that might scare off potential users who don't want to risk a license mistake causing serious trouble *years* from now.

LarsKumbier, 2022-01-28:
Fair enough, thanks for the explanation.

closed 2022-01-28 by LarsKumbier