micropython/micropython-lib #485: Aioble - Discover duplicated or unrelated descriptors

steveli777, 2022-02-24:
When I tried to discover the descriptors, I found the return descriptors are incorrect, I am based on the master branch.
Attached is a modified "temp_client.py" file, I tried to get the descriptor on service (0x1805) and characteristic (0x2a2b). But I got many unrelated descriptors. For example:

- 0x2803
- 0x2800
- 0x2a0f
- 0x2803

[test_aioble.zip](https://github.com/micropython/micropython-lib/files/8137228/test_aioble.zip)

jimmo, 2022-08-05:
@steveli777  I have a fix for this which I will try and get into MicroPython v1.20.

The issue is in the way characteristic discovery works in the BLE spec. When you query for characteristics, it only gives you the start handle of each, and it's up to you to infer the end handle.

NimBLE does not do this for us (although btstack does), and in the case where you're not querying every characteristic (e.g. a uuid filter) you do not have enough information at the layer above to do this inference.

Fortunately (?) though, as per the spec, filtering-by-uuid is actually implemented entirely host-side (!!!) so there's actually no benefit to getting NimBLE to do the uuid filtering and we can do that in MicroPython's bindings instead and then implement the end-handle inference ourselves. This might actually reduce code size a bit.

The next step is to provide a way to pass this to Python, and I think what we'll do is replace the existing definition handle ("def_handle") that is passed to the IRQ and replace that with the end handle. (The definition handle is not useful for anything, none of our other APIs use it).

jonnor, 2024-08-25:
I believe this was fixed in the above merge request by @jimmo . Proposing to close this issue.

closed 2024-08-27 by projectgus