micropython/micropython-lib #510: Interest in a ujwt library?

miguelgrinberg, 2022-07-25:
I have written a tiny version of pyjwt, which I plan to use to handle web sessions in my Microdot library. Would there be any interest in adding it to python-ecosys here? Code is [here](https://gist.github.com/miguelgrinberg/400955ef03169d0a523fb1a2bd133a77) for those interested.

miguelgrinberg, 2022-07-31:
@jimmo will this be reviewed and (hopefully) merged soon? If the library is going to stay parked in this PR for a long time I'd rather publish it on my own, as this is a dependency to one of my projects. Please let me know what your plans are. Thank you.

jimmo, 2022-08-05:
Thanks @miguelgrinberg sorry have been sick for a couple of weeks, looking now.

dpgeorge, 2022-10-04:
The ujwt library was merged, see above PR.

closed 2022-10-04 by dpgeorge