micropython/micropython-lib #512: Since no unsubscribe feature?  What happens to the subscriptions that are orphaned?

Jibun-no-Kage, 2022-07-27:
Since no unsubscribe feature, in simple or robust implementation?  What happens to the subscriptions that are orphaned?  Is the broker left with subscriptions that are bogus?  Does not seem like an elegant or graceful design?

andrewleech, 2022-07-27:
Which library / package are you referring to here? It sounds like one of the mqtt libraries perhaps?
Many packages here are aiming for minimal size as much as robust - but improvements are always welcome!

Jibun-no-Kage, 2022-07-27:
I am using umqtt.simple at the moment, which yes is pretty limited but the robust variant does not seem to have an unsubscribe method either.  I am working on a Pico W, but also have ESP modules I plan to run MicroPython on.

andrewleech, 2022-07-27:
So this is basically a feature request to add unsubscribe functionality to umqtt.simple. 

The most common ethos here is that individuals add just the features they need to modules, as such it seems like none of the previous contributors to that module have needed unsubscribe before! I haven't used that module before myself so don't know any more details about it really.

If you're interested in adding this feature a PR is always welcome.