micropython/micropython-lib #519: aioble char.read max len 22

puppet13th, 2022-08-24:
aioble char.read always return max 22 bytes.
example code:
```
async with connection:
    service = await ble.discover_services(connection,uuid=key_service)
    chars = await ble.discover_chars(service)
    for char in chars:
        char_data = await char.read()
        char_data_hex = hexlify(char_data)
        print(f"char uuid       : {char.uuid}")
        print(f"char data bytes : {char_data} len:{len(char_data)}")
        print(f"char data hex   : {char_data_hex} len:{len(char_data_hex)}")
```
code output :
```
char uuid       : UUID('2d57d0f1-40d5-8be5-8faa-240cc9c31499')
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c' len:8
char data hex   : b'8f3825b73d10a59c' len:16

char uuid       : UUID('2d57d0f2-40d5-8be5-8faa-240cc9c31499')
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd' len:16
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd' len:32

char uuid       : UUID('2d57d0f3-40d5-8be5-8faa-240cc9c31499')
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd\x05\x13>4F_' len:22
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd05133e34465f' len:44

char uuid       : UUID('2d57d0f4-40d5-8be5-8faa-240cc9c31499')
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd\x05\x13>4F_' len:22
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd05133e34465f' len:44
```
the 3rd len should be 24 and 4th len should be 32. i use android EFR connect apps to setup ble advertiser and set services and characteristic.

i have tried to set rxbuf and mtu value
```
aioble.config(mtu=256)
aioble.config(rxbuf=256)
```
but char.read man len does not increase.is there any way to increase max char.read lenght?

andrewleech, 2022-08-24:
The mtu exchange is likely to be the culprit here.

After a ble connection is made one end (usually the central) must trigger the MTU exchange process which is where both sides agree on the common max value they both support

andrewleech, 2022-08-24:
https://github.com/micropython/micropython-lib/blob/c3f4779002abc27991214174d349916c60756183/micropython/bluetooth/aioble/aioble/device.py#L265

puppet13th, 2022-08-24:
> https://github.com/micropython/micropython-lib/blob/c3f4779002abc27991214174d349916c60756183/micropython/bluetooth/aioble/aioble/device.py#L265

thank you very much for the pointer.

i was able to increase char.read len using connection.mtu_exchange:
```
async with connection:
    await connection.exchange_mtu(mtu=128)
    service = await connection.service(key_service)
    chars = []
    async for char in service.characteristics():
        chars.append(char)
    for char in chars:
        mtu = aioble.config('mtu')
        char_data = await char.read()
        char_data_hex = hexlify(char_data)
        print(f"char uuid       : {char.uuid} mtu:{mtu}")
        print(f"char data bytes : {char_data} len:{len(char_data)}")
        print(f"char data hex   : {char_data_hex} len:{len(char_data_hex)}")
        print('')
```

code output:
```
char uuid       : UUID('2d57d0f4-40d5-8be5-8faa-240cc9c31499') mtu:128
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd\x05\x13>4F_[\x81\xbb\xd8\xed\xbe\x07\xe9\x06j' len:32
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd05133e34465f5b81bbd8edbe07e9066a' len:64

char uuid       : UUID('2d57d0f3-40d5-8be5-8faa-240cc9c31499') mtu:128
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd\x05\x13>4F_[\x81' len:24
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd05133e34465f5b81' len:48

char uuid       : UUID('2d57d0f2-40d5-8be5-8faa-240cc9c31499') mtu:128
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c\x80\x9b\xc0$\x16H\xc3\xbd' len:16
char data hex   : b'8f3825b73d10a59c809bc0241648c3bd' len:32

char uuid       : UUID('2d57d0f1-40d5-8be5-8faa-240cc9c31499') mtu:128
char data bytes : b'\x8f8%\xb7=\x10\xa5\x9c' len:8
char data hex   : b'8f3825b73d10a59c' len:16
```

closed 2022-08-24 by jimmo