micropython/micropython-lib #522: aioble exchange mtu sometimes stuck

puppet13th, 2022-09-01:
sometimes aioble exchange mtu stuck without success or timeout.
example code:
```
from time import ticks_ms
ticks_zero = ticks_ms()

import aioble
import uasyncio as asyncio
from ubluetooth import UUID
from random import randint

service = UUID(0x1815)

def timestamp():
    return ticks_ms() - ticks_zero

async def uuid_scan(timeout=2000):
    async with aioble.scan(timeout, interval_us=16000, window_us=16000, active=True) as scanner:
        async for result in scanner:
            if service in result.services():
                return result.device
    return None

async def time_stamp():
    while True:
        await asyncio.sleep_ms(8000)
        print('running time:',timestamp())

async def main_loop():
    while True:
        device = await uuid_scan()
        if device is not None:
            try:
                connection = await device.connect()
            except asyncio.TimeoutError:
                connection = None
            if connection is not None:
                print(timestamp(),device,'connected')
                async with connection:
                    mtu = randint(32,128)
                    print(timestamp(),'exchange mtu',mtu)
                    await connection.exchange_mtu(mtu=mtu)
                    print(timestamp(),'mtu exchanged',aioble.config('mtu'))
                    await asyncio.sleep_ms(randint(200,500))
                print(timestamp(),device,'disconnected')
            else:
                print(timestamp(),'connection failed')
        else:
            print(timestamp(),'device not found')
        await asyncio.sleep_ms(randint(200,500))
        print('\n')
        
asyncio.create_task(time_stamp())
asyncio.run(main_loop())
```

code output:
```
38590 Device(ADDR_RANDOM, 66:9a:9b:5b:d4:a9, CONNECTED) connected
38591 exchange mtu 120
38689 mtu exchanged 120
39089 Device(ADDR_RANDOM, 66:9a:9b:5b:d4:a9) disconnected

39990 Device(ADDR_RANDOM, 66:9a:9b:5b:d4:a9, CONNECTED) connected
39991 exchange mtu 122
40139 mtu exchanged 122
40640 Device(ADDR_RANDOM, 66:9a:9b:5b:d4:a9) disconnected

running time: 41228
41420 Device(ADDR_RANDOM, 66:9a:9b:5b:d4:a9, CONNECTED) connected
41421 exchange mtu 98 <<== exchange mtu stuck
running time: 49239
running time: 57248
running time: 65258
running time: 73268
```

andrewleech, 2022-09-01:
I've added a timeout arg in https://github.com/micropython/micropython-lib/pull/465

puppet13th, 2022-09-01:
After 10 months and not merge yet. I will try it and report back.

puppet13th, 2022-09-01:
#465 confirmed fixed the problem.
asyncio.TimeoutError raised if timeout occurred when doing exchange_mtu method.

closed 2022-09-01 by puppet13th

jimmo, 2022-09-01:
Sorry about the delay on this -- I've merged https://github.com/micropython/micropython-lib/pull/465