micropython/micropython-lib #535: Cannot install aioble lib in Raspberry Pi Pico W.

shbobur, 2022-09-15:
I am running MicroPython v1.19.1 on Pico W.

When I try to install package aioble through minicom I get this error:

```
>>> upip.install('micropython-aioble')\
   Installing to: /lib/
   Warning: micropython.org SSL certificate is not validated
   Error installing 'micropython-aioble': Package not found, packages may be partially installed
```

And if I install it without word `micropython` I get this result:

```
>>> upip.install('aioble')
   Installing to: /lib/
   Warning: micropython.org SSL certificate is not validated
   Installing aioble 0.0.1 from 
   https://files.pythonhosted.org/packages/08/28/d966a681939c373bee078e0afb9afe776cf1739dd0ee5e76d12a838e1825/aioble-0.0.1-py3.6.egg
   Error installing 'aioble': compression header, packages may be partially installed
```

I know in both cases errors are self descriptive. But what can I do in these case?

andrewleech, 2022-09-15:
The Raspberry Pi Pico W currently does not have support for Bluetooth.
It's expected to be added at some point in the future, but I have no information when this might occur.

Also, currently `upip` does not have support for installing directly from micropython-lib, it's an old tool that can only see things that were pushed to pypi or a legacy micropython package host. 

An update to this tool to work with the new manifest system which is being build and merged incrementally lately is planned, but there's not ETA on this yet. Once that's done though all the packages in this repo will be available for install.

In the mean time packages can be copied manually from this repo onto a device with `mpremote`.

shbobur, 2022-09-15:
@andrewleech Thank you very much for your detailed answer.

closed 2022-09-15 by shbobur