micropython/micropython-lib #538: urequests TLS requests fail on ESP32 with ESP-IDF 4.3 and 4.4

chaseadam, 2022-09-20:
TLS calls work in the 1.19.1 official release binary and with master commit 13c4470fd04981fce3fd29771e1d973febbe7d60 built against ESP-IDF 4.2.

```
MicroPython v1.19.1-437-g13c4470fd on 2022-09-20; ESP32 module with ESP32
Type "help()" for more information.
>>> gc.mem_free()
29712
>>> gc.collect()
>>> gc.mem_free()
69664
>>> r = requests.get('https://google.com')
>>>
```

Using **ESP-IDF 4.3 and 4.4** with 1.19.1 sources or HEAD of master result in what appear to be memory issues.

Output:

```
>>> import urequests as requests
>>> gc.mem_free()
23360
>>> gc.collect()
>>> gc.mem_free()
66336
>>> r = requests.get('https://google.com')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 180, in get
  File "urequests.py", line 93, in request
OSError: 16
```

I have also seen the following errors:

This is most common
```
Error: [Errno 113] ECONNABORTED, Reason: null
```

I saw this intermittently while testing
```
OSError: (-17040, 'MBEDTLS_ERR_RSA_PUBLIC_FAILED+MBEDTLS_ERR_MPI_ALLOC_FAILED')
```

jimmo, 2022-09-21:
@chaseadam this is https://github.com/micropython/micropython/issues/8940

(And unfortunately, it's the reason why we still do ESP32 builds with IDF 4.2)

chaseadam, 2022-10-12:
closing out as this is duplicate of https://github.com/micropython/micropython/issues/8940

closed 2022-10-12 by chaseadam