micropython/micropython-lib #548: uJSON indent

meirarmon, 2022-10-12:
Hi,
In my project, I would like the of dumping json with `indent`. If I implemented it, would it be something that would be merged?

jimmo, 2022-10-13:
> Hi, In my project, I would like the of dumping json with `indent`. If I implemented it, would it be something that would be merged?

Not sure... it's extra code size for something that is pretty marginal in its utility (i.e. it's always very easy to re-format JSON before viewing... I can imagine a case maybe where you're generating a config file or something that needs to be human editable?). A rudimentary json-indenter could also be written in Python without too much difficulty.

FWIW, we provide a more full-featured `json` implementation in micropython-lib https://github.com/micropython/micropython-lib/tree/master/python-stdlib/json (although it's quite big!).

meirarmon, 2022-10-13:
Yes, it is very big. I was just thinking that implementing `indent` in C like th rest of uJson should not increase the size too much.
But if you think it is too marginal, I will leave this issue

jimmo, 2022-10-14:
> Yes, it is very big. I was just thinking that implementing `indent` in C like th rest of uJson should not increase the size too much. But if you think it is too marginal, I will leave this issue

Unfortunately with these things the only way to find out is to implement it and measure. But I can't see it being under a few hundred bytes.

Honestly I think a Python post-formatter would be the way to go.

closed 2022-10-14 by meirarmon