micropython/micropython-lib #567: Unable to Publish Message to AWS IoT with Micropython v1.19

vivekmystrey, 2022-11-07:
With the following test script, I am able to publish messages to AWS Cloud for MicroPython version v1.17. However, the same script does not work for MicroPython version 1.19 and it errors out. The issue is replicable for both ESP32-S3 and ESP32-C3.

Test Script:

```
#AWS MQTT client cert example for esp8266 or esp32 running MicroPython 1.9
from umqtt.robust import MQTTClient
import time
import network
from time import sleep
import usocket as socket
import ujson

#Certs for ESP8266
CERT_FILE = "/flash/client.crt"
KEY_FILE = "/flash/client.key"

#if you change the ClientId make sure update AWS policy
MQTT_CLIENT_ID = "basicPubSub"
MQTT_PORT = 8883

MQTT_KEEPALIVE_INTERVAL = 45
THING_NAME = "ESP32-Core1"
SHADOW_GET_TOPIC = "$aws/things/" + THING_NAME + "/shadow/get"
#SHADOW_UPDATE_TOPIC = "$aws/things/" + THING_NAME + "/shadow/update"
SHADOW_UPDATE_TOPIC = "sdk/test/Python"

RESPONSE_RECEIVED = False

#if you change the topic make sure update AWS policy
MQTT_TOPIC = "esp8266-topic"

#Change the following three settings to match your environment
MQTT_HOST = "*****************-ats.iot.us-east-1.amazonaws.com"
WIFI_SSID = "***************"
WIFI_PW = "*************"

mqtt_client = None

def sub_cb(topic, msg, retained, duplicate):
    print((topic, msg, retained, duplicate))

def connect_mqtt():
    try:
        with open(KEY_FILE, "r") as f:
            key = f.read()

        print("MQTT received KEY")

        with open(CERT_FILE, "r") as f:
            cert = f.read()

        print("MQTT received CERTIFICATE")
        mqtt_client = MQTTClient(client_id=MQTT_CLIENT_ID, server=MQTT_HOST, port=MQTT_PORT, keepalive=5000, ssl=True, ssl_params={"cert":cert, "key":key, "server_side":False})
        mqtt_client.connect()
        mqtt_client.set_callback(sub_cb)
        print("MQTT is connected")
        a = {"message": "Hello from IoT console"}
        try:
            a = mqtt_client.publish(SHADOW_UPDATE_TOPIC, ujson.dumps(a))
            print("Publish Message", a)
        except Exception as e:
            print('Cannot Publish' + str(e))
            #mqtt_client.wait_msg()

    except Exception as e:
        print('Cannot connect to MQTT ' + str(e))
        raise

def connect_wifi(ssid, pw):
    wlan = network.WLAN(network.STA_IF)

    if(wlan.isconnected()):
        wlan.disconnect()
    nets = wlan.scan()

    if not wlan.isconnected():

        wlan.active(True)
        wlan.connect(WIFI_SSID, WIFI_PW)
        while not wlan.isconnected():
            pass
    print("Connected as:", wlan.ifconfig())

try:
    print("Connecting to WIFI...")
    connect_wifi(WIFI_SSID, WIFI_PW)
    print("Connecting to MQTT...")
    connect_mqtt()
    print("OK")

except for Exception as e:
    print(str(e))

```

Error Message:

```
>>> %Run -c $EDITOR_CONTENT
Connecting to WIFI...
Connected as: ('***********', '**************', '************', '***********')
Connecting to MQTT...
MQTT received KEY
MQTT received CERTIFICATE
Cannot connect to MQTT bytes index out of range
bytes index out of range

```

Can you please check and advise, if the parameters to setup Mqtt connection are right?

vivekmystrey, 2022-11-07:
And further debugging, found with MicroPython Version v1.19 at line 96 (`resp = self.sock.read(4)`) in simple.py inside connect function it is getting null response whereas with MicroPython Version 1.17 it gets byte values.

chaudhariatul, 2023-02-09:
Certificate and Key files need to be in der format

```
openssl x509 -in client.crt -out client_crt.der -outform DER
openssl rsa -in client.key -out client_key.der -outform DER
```

erinlkolp, 2023-04-15:
Have there been any updates on this, @vivekmystrey ? Did you get it working? I'm hitting the same error.

babakc, 2023-06-18:
It's likely your certificate policy is too restrictive. Make sure you have iot:connect permissions set correctly.