micropython/micropython-lib #604: ntptime: Does not handle KoD (Kiss of Death) resonse.

peterhinch, 2023-02-01:
A call to `.time()` can return a large negative number which results in a rather confusing exception in `.settime()`. This can occur if an application issues NTP requests too frequently: see [this post](https://github.com/micropython/micropython/discussions/10611#discussioncomment-4812100). I suspect there may also be other circumstances in which an NTP query returns zero.

Another problem reported in discussions is that `.time()` frequently times out. Clearly the timeout must be short, but this implies that on many networks timeouts will be common.

There are a number of simple fixes. One is for `.time()` to  return an error value in either of these cases which is tested by `.settime()`. The latter could return a `bool` indicating success or failure.

massimosala, 2023-06-01:
Hi Peter

I dislike the ntptime module, I think it is like "how to confuse a cat" !

Probably a lot of users get KoD because they don't realize that ntptime.time() isn't time.time() ...

I wrote my version, with more strict checking about the servers reponse.

Feel free to read and use it.
Ah, you have to remove the _compatibility_ module and the gc_collect()
The _compatibility_ is just to use this module also on windows / linux, defining _const_

Eventually I can donate it to mp ecosystem.

[ntptime2.py.txt](https://github.com/micropython/micropython-lib/files/11628081/ntptime2.py.txt)

massimosala, 2023-06-01:
I see now I use a variable MICRO in settime()

Just replace the code with:

```
	try :
		from machine import RTC
		RTC().datetime((tm[0], tm[1], tm[2], tm[6] + 1, tm[3], tm[4], tm[5], 0))
	except :
		print("Unsupported platform, no RTC")
```

and the printf() with print()

Ciao

closed 2026-09-14 by dpgeorge