micropython/micropython-lib #606: Package installation issue on Seeed XIAO SAMD21

vova-lantsov-dev, 2023-02-02:
> MicroPython v1.19.1-850-gfe2a8332f on 2023-02-01; Seeed Xiao with SAMD21G18A
> Also tried a stable 1.19.1

Hi. I have a need to make http requests repeatedly and these are the imports I'm trying to do:

```
import socket as usocket
from network import WLAN
import ssl as ussl
import time
import machine
```

2 packages are built-in (time, machine), but other 3 packages are external.
I'm unable to install them.

What I already tried:
1) installing packages using `Tools` - `Manage packages...` in Thonny IDE
This window shows `Package manager is not available for this interpreter (no absolute lib directory in sys.path)`
I tried to set the `sys.path` as well from the shell, but no luck.
2) using `mip` from shell gives a `Module not found` error
3) using `mpremote mip install --target lib/ {package_name}` works well, but the main problem is that this package can't install `socket` and `network` packages
`mpremote mip install --target lib/ socket` gives the following error:
```
Install socket
Installing socket (latest) from https://micropython.org/pi/v2 to lib/
Package may be partially installed
mpremote: Package not found: https://micropython.org/pi/v2/package/6/socket/latest.json
```
There is no `network` package as well.
But PyPi has the corresponding packages called `micropython-ssl`, `micropython-network`, `micropython-socket` that I can't install via `mip` tool.
4) I also tried to download a `.zip` archive from PyPi and manually move the library into device\`s `lib` directory. Code is still not able to found this dependency with `Module not found` error.

andrewleech, 2023-02-02:
Unfortunately those packages on pypi are legacy things from a previous micropython contributor that don't work - if you see the description on eg. https://pypi.org/project/micropython-ssl/

> This is a dummy implementation of a module for MicroPython standard library. It contains zero or very little functionality, and primarily intended to avoid import errors

It's a shame there are a bunch of dummy packages that do nothing but confuse users, bit this is one of the reasons `mip` doesn't use pypi at all, it uses a new system that's based directly on micropython-lib.

These particular packages; socket, SSL, etc. do need to be built-in modules, and typically are on boards with native networking support.

Most importantly, the Seeed Xiao with SAMD21G18A is this board isn't it?
https://wiki.seeedstudio.com/Seeeduino-XIAO/

It doesn't seem to have any kind of wifi chip or physical network support at all? That's why these modules can't be available in micropython.

vova-lantsov-dev, 2023-02-02:
Thank you for response 🙂

closed 2023-02-02 by vova-lantsov-dev