micropython/micropython-lib #607: Install gives error code 1

Richard238, 2023-02-05:
Tried Thonny / Tools / Manage Packages / mqtt <search> / 

Found, second in list, but got following error on attempted install

`'--upgrade-strategy', 'only-if-needed', 'micropython-umqtt.simple', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.`

andrewleech, 2023-02-05:
To install packages from here `mip` should be used: https://docs.micropython.org/en/latest/reference/packages.html

Currently it's only available in the nightly builds, not 19.1, so you might need to update your micropython.

Thonny uses its own install system that doesn't directly relate to this repo, any support requests relating to thonny will need to go to their own forum etc.

jonnor, 2024-08-25:
Issues with installing packages with Thonny must be directed at Thonny.

closed 2024-08-25 by mattytrentini