micropython/micropython-lib #610: NRF24L01 problem with failing send command

vendetta1987, 2023-02-09:
I'm trying to communicate between an RPi4 running python and the [py-nrf24](https://github.com/bjarne-hansen/py-nrf24) lib and a Pi Pico using the micropython NRF24L01 driver.
I can transmit data and started building my application but am encountering intermittent data transfer problems. I started out with the given examples on both sides and cut them down step by step to fit my needs.
It all boils down to the following line which I first ignored:
https://github.com/micropython/micropython-lib/blob/a5ef231e7d854bab0e6db95f1676173e2281ee3b/micropython/drivers/radio/nrf24l01/nrf24l01test.py#L63 Putting a print there tells me that the send command before always succeeds ones and then fails on the next call.
I then tried some variations of waiting (maybe it just needs time to send and acknowledge?) but ended up simply using https://github.com/micropython/micropython-lib/blob/a5ef231e7d854bab0e6db95f1676173e2281ee3b/micropython/drivers/radio/nrf24l01/nrf24l01.py#L227 This works most of the time even without waiting for any acknowledgement.
Calling https://github.com/micropython/micropython-lib/blob/a5ef231e7d854bab0e6db95f1676173e2281ee3b/micropython/drivers/radio/nrf24l01/nrf24l01.py#L245 immediately after send_start() like send() does definitely seems fishy on the Pico, some timing issue or similar.
I'm just sending around a dozen bytes in 100ms intervals for testing. I now ran into, as it seems, some kind of buffer overflow as the pico stops sending after some amount of time.
Currently I'm using these settings on the Pi Pico (and RPi 4 equivalent):
- payload size = 32
- POWER_3
- SPEED_250K
- SPI 0
- breadboard
- USB power from PC
- devices separated by one building story (although the send successful -> send fail -> success -> fail loop happened with both radios next to each other as well)

I didn't yet fully understand the pipes/addresses concept. Both Pis use the same address, the Pico sends, the RPi4 receives. As there seem to be up to 6 receiving pipes/addresses and the RPi lib by default puts my configured address in slot 1 (0 based index) maybe the acknowledgment functionality is the culprit.
If both NRF use the same address for communication, on which pipe/address do they exchange the acknowledgement?
How do you even configure it in the micropython lib?

pavlot, 2023-04-21:
For NRF device pipe is a concept of receiving device. So each device intended to receive any data must have configured at least one pipe with adress. It is possible for one receiver to have up to 6 pipes with different addresses configured. Each transmitter must send data to one of configured addresses.

For acknowledge to work, on transmitter, you have to define at least one pipe with the same address as on receiver. Example:
Receiver: 
```
Pipe 0 addr 1122332211
Pipe 1 addr 2233443322
```

Transmitter:
```
TX_ADDR 2233443322
For ACK Enable pipe 0, set its address to 2233443322 and it will do the trick
```

Note pipe number is not mandatory to be the same on transmitter and receiver, only address matters.

marcsello, 2024-12-30:
> Putting a print there tells me that the send command before always succeeds ones and then fails on the next call.

I had a very similar case. I was using a pi pico, and only every second send send was succeeding, but it was acting quite weird. Only difference was that I wasn't getting any error on the sender side.

Eventually I figured out that for some reason, the driver didn't let enough time for my module to start up. I've opened a PR to fix this. Could you please test if that fixes your problem as well?

https://github.com/micropython/micropython-lib/pull/953