micropython/micropython-lib #611: Problem with dht library

gandipg, 2023-02-12:
Hello
Problem with dht library not working properly in subzero temperatures with dht22 module.
reports (-3264.4°C)
it is written here about a domestic problem  https://forums.raspberrypi.com/viewtopic.php?t=296097

Keelando, 2023-02-28:
Indeed! I'm not well-versed enough to suggest a proper solution at this time but I've just been adding the offset value corresponding to -0.1C and then multiplying by -1

dpgeorge, 2023-03-03:
I tested the `dht` driver with a DHT22 and it works OK for me at temperatures less than 0 Celsius.

The `dht` driver here assumes that the top bit of the returned temperature value is a sign bit.  But if you are getting values like -3264.4 then it could be that your DHT22 returns a 2's complement signed value.

After you do a reading can you print out the buffer, like this:
```py
dht.measure()
print(dht.temperature(), dht.buf)
```
That will help narrow down the problem.

If it is 2's complement then you have a non-standard DHT22 module and you'll need to use conversion like this:
```py
def dht_temp(dht):
    value = dht.buf[2] << 8 | dht.buf[3]
    if value & 0x8000:
        value -= 0x10000
    return value * 0.1
```