micropython/micropython-lib #621: pathlib.Path.__rtruediv__ micropython incompatability.

BrianPugh, 2023-02-24:
I wanted to add the following magic method to the ``pathlib.Path`` object:

```
    def __rtruediv__(self, other):
        return Path(other, self._path)
```

so that the following unittest would be satisfied:

```
    def test_rtruediv(self):
        res = "foo" / Path("bar")
        self.assertTrue(res == Path("foo/bar"))
```

However, I get the following stacktrace:

```
Traceback (most recent call last):
  File "/Users/brianpugh/.micropython/lib/unittest/__init__.py", line 392, in run_one
  File "./tests/test_pathlib.py", line 327, in test_rtruediv
TypeError: can't convert 'Path' object to str implicitly
```

The `__rtruediv__` method never gets called, suggesting that the `str.__truediv__` gets called in a way that is different than in cpython. I'm not familiar enough with cpython or micropython internals to know `str`'s `__truediv__` implementation, but something is up.

dlech, 2023-02-24:
Does your target have `MICROPY_PY_ALL_SPECIAL_METHODS` enabled?

BrianPugh, 2023-02-24:
I no longer see the options in `ports/unix/mpconfigport.h`, but I added it as true. Did a clean rebuild; same issue.

dlech, 2023-02-24:
And also `MICROPY_PY_REVERSE_SPECIAL_METHODS`?

BrianPugh, 2023-02-24:
no luck setting both to true; i'm digging a lil deeper.

BrianPugh, 2023-02-24:
I think this comment here is incorrect:

https://github.com/micropython/micropython/blob/2e4dda3c2072c4de15ac8d51ddb471b9c3833db7/py/objstr.c#L376

I'll see if I can come up with a fix.

BrianPugh, 2023-02-24:
Will be resolved in:

https://github.com/micropython/micropython/pull/10844

Closing this issue for now.

closed 2023-02-24 by BrianPugh

dpgeorge, 2023-05-19:
This (reverse inplace ops with str) is now supported in MicroPython.

BrianPugh, 2023-05-19:
excellent! iirc i have a test somewhere just to make sure this works as expected for pathlib, I'll open a PR. Thank you very much!

BrianPugh, 2023-05-19:
Added support in https://github.com/micropython/micropython-lib/pull/663