micropython/micropython-lib #626: SDcard save year

Gianfilippo980, 2023-03-05:
Whenever I write a fie to a MicroSD card from my Raspberry Pi Pico, if I open the card in a computer the date in something like 05/03/2103 (current date +80 years).
I am sure that the RTC on the Pico is set correctly (i do write a timestamp in the file, and I tested with it connected to a PC with Thonny which syncs it up). This is not a deal breaker, as I said, I do write a timestamp in the file, but it looks strange.

I downloaded the sdcard.py file a couple of days ago from this repository and my Pico is running MicroPython 1.19.
The simplest code that I tried is: 

`import machine
import sdcard
import uos

# Assign chip select (CS) pin (and start it high)
cs = machine.Pin(17, machine.Pin.OUT)

# Intialize SPI peripheral (start with 1 MHz)
spi = machine.SPI(0,
                  baudrate=1000000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))

# Initialize SD card
sd = sdcard.SDCard(spi, cs)

# Mount filesystem
vfs = uos.VfsFat(sd)
uos.mount(vfs, "/sd")

try:
    open("/sd/non.txt", "r")
except:
    print ("Non c'è")
else:
    print ("C'è")

# Create a file and write something to it
with open("/sd/test01.txt", "w") as file:
    file.write("Hello, SD World!\r\n")
    file.write("This is a test\r\n")

# Open the file we just created and read from it
with open("/sd/test01.txt", "r") as file:
    data = file.read()
    print(data)`

dpgeorge, 2023-03-07:
I tested a Pico W with USB MSC enabled, and a VfsFat filesystem.  Writing to a file from the Pico W, and then viewing the file on the PC, the date is correct.

Can you run `machine.RTC().datetime()` before opening the file, to see exactly what datetime it is using for the file's timestamp?

Gianfilippo980, 2023-03-07:
Well, yes, my first suspect was that I somehow had set the RTC wrong, but, as you can see... it isn't.
[image]
[image]
I also opened the SD from Raspberry, just in case this was linked to Windows, but it has the same issue.
Again, this is not really too big of an issue for me, it just looks odd.

dpgeorge, 2023-03-07:
Oh, I see you are using MicroPython v1.19.

I think your issue is exactly the same as https://github.com/micropython/micropython/issues/9535 and has been fixed in the firmware in https://github.com/micropython/micropython/commit/92d911803812100fac109b6108ab005e7bbd4c82

So please try the latest "unstable" release of firmware, that should fix the issue.

closed 2023-03-07 by dpgeorge