micropython/micropython-lib #627: no SD card

hairem, 2023-03-06:
I am trying to use this library on a rpi pico:
cs = machine.Pin(22, machine.Pin.OUT)
spi = machine.SPI(0,
                  baudrate=1000000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))
sd = sdcard.SDCard(spi, cs)

# Mount filesystem
vfs = uos.VfsFat(sd)
os.mount(vfs, "/sd")
but every time I try to run my script in micropython I get eh error of:
"Traceback (most recent call last):
  File "<stdin>", line 27, in <module>
  File "lib/sdcard.py", line 54, in __init__
  File "lib/sdcard.py", line 82, in init_card
OSError: no SD card"

Any ideas on how to fix this?

hairem, 2023-03-06:
I have the SDcard formatted in Fat32, but it doesn't seem to get to that point, it's not mounting the card because it can't see the card

dpgeorge, 2023-03-07:
The most likely explanation is that the SD card is not wired up correctly (has it got power?).

hairem, 2023-03-07:
Yes, I have used both my 5v rail and my 3.3v rail to power the unit, at different times since I thought it might not be receiving enough power on the 3.3v.  When on the 5v rail I do get a different error:
Traceback (most recent call last):
  File "<stdin>", line 27, in <module>
  File "lib/sdcard.py", line 54, in __init__
  File "lib/sdcard.py", line 98, in init_card
  File "lib/sdcard.py", line 196, in readinto
OSError: timeout waiting for response

hairem, 2023-03-13:
I confirmed power and I am still having issues with accessing the SD card. Any suggestions?

hairem, 2023-03-13:
'''
This is version 0.9a of the PM Monitor program. Data is stored a SD Card.
'''
import machine
from machine import Pin,UART
import time
import bme280
from lib import RTC_DS3231
from pms5003 import PMS5003
from lib import sdcard
import uos
import gc
gc.enable()

#Setup SPI SD Card
#assign CS Pin
cs = machine.Pin(22, machine.Pin.OUT)
spi = machine.SPI(0,
                  baudrate=1000000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))
sd = sdcard.SDCard(spi, cs)

# Mount filesystem
vfs = uos.VfsFat(sd)
uos.mount(vfs, "/sd")

#I2C device Settings
#i2c0=machine.I2C(1,sda=machine.Pin(8), scl=machine.Pin(9), freq=400000)
i2c0=machine.I2C(0,sda=machine.Pin(0), scl=machine.Pin(1), freq=400000)

#Serial Ports
uart1 = UART(0, baudrate=9600, tx=Pin(12), rx=Pin(13))
uart1.init(bits=8, parity=None, stop=1)
uart0 = UART(0, baudrate=9600, tx=Pin(16), rx=Pin(17))
uart0.init(bits=8, parity=None, stop=1)

#Device settings
bme = bme280.BME280(i2c=i2c0)
rtc = RTC_DS3231.RTC()
pms5003 = PMS5003(
    uart=uart1,
    pin_enable=machine.Pin(3),
    pin_reset=machine.Pin(2),
    mode="active"
)

#Clean up
def CleanMet(MET):
 Met=str(MET)
 Met=str(Met.replace("(",",").replace(")",","))
 met=str(Met.replace("'","").replace("C","").replace("hPa","").replace("%",""))
 met = met.split(",")
 return(met)

#Values
def Readings():
 try:
  i = 0
  Temp = 0.0
  Press = 0.0
  RH = 0.0
  PM25 = 0.0
  while 1: 
   t = rtc.DS3231_ReadTime(2)
   MET = bme.values
   Met = CleanMet(MET)
   data = pms5003.read()
   Data = data.pm_ug_per_m3(2.5)
   #print(str(t) + "," + str(Met[1]) + "," + str(Met[2])+ "," + str(Met[3]) + "," +str(Data))
   #print(rtc.DS3231_ReadTime(3))
   if int(rtc.DS3231_ReadTime(3)) > 0:
     Temp = float(Temp) + float(Met[1])
     Press = float(Press) + float(Met[2])
     RH = float(RH) + float(Met[3])
     PM25 = float(PM25) + float(Data)
     i=i+1
   else:
     avgT = round(Temp/i,1) 
     avgP = round((Press/i)*0.75006,1)
     avgRH = round(RH/i, 1)
     avgPM25 = round(PM25/i,1)
     line = str(t) + "," + str(avgT) + "," + str(avgP) + "," + str(avgRH) + "," + str(avgPM25)+ "," + str(i)
     print(str(t) + "," + str(avgT) + "," + str(avgP) + "," + str(avgRH) + "," + str(avgPM25)+ "," + str(i))
     with open("/sd/PMData.csv", "a") as file:
      file.write(str(line) + "\r\n")
      file.close()
     #print(gc.mem_free())
     i = 0
     Temp = 0.0
     Press = 0.0
     RH = 0.0
     PM25 = 0.0
     gc.collect()
 except:
     print("Error")
     gc.collect()
     time.sleep(1)
     Readings()

if __name__ == "__main__":
 Readings()

andrewleech, 2023-03-13:
If 5v was applied to the card it might be dead now, unless your card reader is on a breakout with level shifters? Can you confirm the card still works in a PC reader? Are all the other spi pins definitely wired to the correct sd pins?

hairem, 2023-03-13:
It's still good because I was using it today to move files from one
computer to another

andrewleech, 2023-03-13:
Pinout: are all the card reader pins wired correctly?
Card reader: is the socket faulty?
Software: where did your copy of sdcard.py come from?

https://core-electronics.com.au/guides/raspberry-pi-pico/makerverse-micro-sd-adapter-micropython-guide/

You might also have more luck asking about this on the micropython discord, this issue tracker is generally for tracking SW bugs and not monitored by many people.

hairem, 2023-03-13:
I have tried multiple SD card reader modules, same error, I have confirmed
the wiring multiple times and had a second person check it. I got the
library from this GitHub.

andrewleech, 2023-03-13:
Might be an issue with the size / brand of sd card, seems there can be some differences at times: https://e2e.ti.com/support/microcontrollers/msp-low-power-microcontrollers-group/msp430/f/msp-low-power-microcontroller-forum/294813/sdhc-code-with-spi-works-on-kingston-card-but-not-on-sony-card

hairem, 2023-03-13:
I doubt it's the SD card because I have used this one before with this spi
module

SanDisk Industrial 8GB Micro SD...
https://www.amazon.com/dp/B085GL8XBH?ref=ppx_pop_mob_ap_share

andrewleech, 2023-03-13:
I haven't used this sdcard.py on a pico but there's plenty of tutorials online using it so it must be broadly compatible.

I keep going back to the pinout because it's a common issue and we can't actually see what you've done; so for the pins specified in the code, eg. Pin(18) have you definitely wired GP18 and not pin number 18?

Also, what version micropython are you running on the pico? Maybe try updating that to latest nightly if you haven't already recently.

hairem, 2023-03-15:
(sysname='rp2', nodename='rp2', release='1.19.1', version='v1.19.1 on 2022-06-18 (GNU 11.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')

I have multiple other modules that work fine, the only issue I am having is with the SD card module. I have looked at several tutorials and walkthroughs and have tried them as well, well the ones that don't link to a dead copy of this library, The first one I tried was the Digikey walkthrough but I run into the same issue which leads me to ask the owner of the GitHub.

hairem, 2023-03-15:
I am using the following pins:
pin 21 = GP16 =MISO
pin 22 = GP17 = CS
pin 24 = GP18 = SCK
pin 25 = GP19 = MOSI

andrewleech, 2023-03-15:
That looks like the issue then!
In your code above you have:
```
cs = machine.Pin(22)
sck = machine.Pin(18)
mosi = machine.Pin(19)
miso = machine.Pin(16)
```
The Pin number refers to the GP number, not the raw pin number. It looks like you have the Pins that are passed to Spi correct but cs is declared as GP22, not GP17

hairem, 2023-03-15:
That was fixed and still getting the same error

khansahab117, 2023-12-24:
> Yes, I have used both my 5v rail and my 3.3v rail to power the unit, at different times since I thought it might not be receiving enough power on the 3.3v. When on the 5v rail I do get a different error: Traceback (most recent call last): File "", line 27, in File "lib/sdcard.py", line 54, in **init** File "lib/sdcard.py", line 98, in init_card File "lib/sdcard.py", line 196, in readinto OSError: timeout waiting for response

I am also getting this error. Were you able to find the fix for this?

hairem, 2023-12-24:
I did but but I would have to be at my desk to look at what I did to fix it
and I am in vacation with family ATM.

fox-Nh133, 2024-02-09:
I also encountered this error; ```"OSError: no SD card"```.
for me, adjusting ```_CMD_TIMEOUT``` value in sdcard.py from 100 to 1000, and connecting sufficient power to the sd module solved the problem.

Initially, I wired 3.3v power to the sd module, but after checking the datasheet, it turned out that the module accepts 4.5-5.5v as input. so I switched the power supply port from ```VSYS``` to ```VBUS```.

To check input power of sd module might help.

Hristo-Nikodimov-Nenkov, 2024-03-02:
Same problem, both with Pico W and RP2040 clone with 16MB ROM.

from machine import SPI
from sdcard import SDCard
from uos import VfsFat, mount, listdir
cs_pin = Pin(13, Pin.OUT)
sck = Pin(14)
mosi = Pin(15)
miso = Pin(12)

spi = machine.SPI(1,
                  baudrate=1000000,
                  polarity=0,
                  phase=0,
                  bits=8,
                  firstbit=SPI.MSB,
                  sck=sck,
                  mosi=mosi,
                  miso=miso)

sd = SDCard(spi, cs_pin)
vfs = VfsFat(sd)

os.mount(vfs, '/card')

print(listdir("/card"))

>>>
Traceback (most recent call last):
  File "<stdin>", line 50, in <module>
  File "/lib/sdcard.py", line 54, in __init__
  File "/lib/sdcard.py", line 82, in init_card
OSError: no SD card

I've tried with 1GB Nokia, 2GB SanDisk and 4GB N/A all cards are formated as FAT32 and work fine on both my laptop and my headphones,

As for power I've used both the 3.3V of the Picos and external power converter for breadbords.

dev12p, 2024-03-22:
> Same problem, both with Pico W and RP2040 clone with 16MB ROM.
> 
> from machine import SPI from sdcard import SDCard from uos import VfsFat, mount, listdir
> 
> cs_pin = Pin(13, Pin.OUT) sck = Pin(14) mosi = Pin(15) miso = Pin(12)
> 
> spi = SPI(1, baudrate=1000000, polarity=0, phase=0, bits=8, firstbit=SPI.MSB, sck=sck, mosi=mosi, miso=miso)
> 
> sd = SDCard(spi, cs_pin) vfs = VfsFat(sd)
> 
> mount(vfs, '/card')
> 
> print(listdir("/card"))
> 
> > > >
> 
> Traceback (most recent call last): File "", line 50, in File "/lib/sdcard.py", line 54, in **init** File "/lib/sdcard.py", line 82, in init_card OSError: no SD card
> 
> I've tried with 1GB Nokia, 2GB SanDisk and 4GB N/A all cards are formated as FAT32 and work fine on both my laptop and my headphones,
> 
> As for power I've used both the 3.3V of the Picos and external power converter for breadbords.
> 
> I've even tried the 1GB and 2GB cards formated as FAT with same result.

Did you end up finding a solution?

hairem, 2024-03-22:
Went with a different SD card module

Hristo-Nikodimov-Nenkov, 2024-04-02:
I found the problem, it was a combination of old breadboard and cheep patch cables. 
I'm realy sorry for not even suspecting the breadboard.

ZeynX92, 2024-06-05:
Hello, I Have same problem

```py
import machine
import sdcard
import uos

cs = machine.Pin(1, machine.Pin.OUT)

spi = machine.SPI(0,
          baudrate=1000000,
          polarity=0,
          phase=0,
          bits=8,
          firstbit=machine.SPI.MSB,
          sck=machine.Pin(2),
          mosi=machine.Pin(3),
          miso=machine.Pin(4))

sd = sdcard.SDCard(spi, cs)

vfs=uos.VfsFat(sd)
uos.mount(sd,'/sd')
```

Error
```
Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
  File "sdcard.py", line 54, in __init__
  File "sdcard.py", line 82, in init_card
OSError: no SD card
```
I really don't know what to do. because i check all contacts, power, use and format 3 different SD cards, change `_CMD_TIMEOUT`...

Hiroyugane, 2024-06-10:
> I found the problem, it was a combination of old breadboard and cheep patch cables. I'm realy sorry for not even suspecting the breadboard.

Can confirm. Had the same problem and tried reformatting the sd card several times. 
Turns out, my SD Reader breakout board pins were too short and did not quite reach into the breadboard - therefore not making a good connection.
Used Pin Contact spray and other patch cables. Solved the problem for me.

Thanks for commenting this!

slabua, 2024-07-16:
Following :\

ZeynX92, 2024-07-16:
My story with this error: https://github.com/micropython/micropython-lib/issues/871

slabua, 2024-07-16:
I might have found something!

```
spi = SPI(
    id=1,
    sck=Pin(10, Pin.OUT),
    mosi=Pin(11, Pin.OUT),
    miso=Pin(8, Pin.OUT),
)
```

It seems like some other pins are being bullied by the lib.
These are the first listed pins for SPI1 and there is no error.
Might as well work for the first listed SPI0 pins.

Also, if the sdcard contains chinese characters, generally some issue with unicode, it doesn't work.

KarthikDani, 2024-12-20:
> I also encountered this error; `"OSError: no SD card"`. for me, adjusting `_CMD_TIMEOUT` value in sdcard.py from 100 to 1000, and connecting sufficient power to the sd module solved the problem. [...]

After hours of checking different configuration and stuff, this indeed worked! Thanks a lot!