micropython/micropython-lib #638: umqtt.simple not working with Mosquitto broker 2.0.15

YeeMoone, 2023-04-04:
Hi!
umqtt.simple fails to connect to mosquito 2.0.15. It throws back MQTTException: 2.
My mosquitto log without any errors. The logs displayed are the same as when connecting to the server using the MQTT debugger(An app), so, maybe my mpy device is connected to mosquiro broker? But how to fix it?

By the way, my MQTT debugger was able to successfully connect to the mosquito without any errors, including publish and subscribe messages

my device: nodemcu-8266(12F) from Espressif
my mqtt server: Mosquitto broker 6.2.0 in homeassistant(VMware)

basta, 2023-04-04:
Try setting  keepalive=60 as an argument for MQTTClient, that fixed it for me

YeeMoone, 2023-04-05:
thanks, it's working for me too

closed 2023-04-05 by YeeMoone

andrewleech, 2023-04-05:
As Mosquitto is very common, maybe it makes sense to raise the default timeout to that?

basta, 2023-04-05:
Afaik it's caused by an update to mqtt protocol that requires this argument to be specified. It's not related to the default value. But I don't understand it much