micropython/micropython-lib #639: urequests timeout

TaylorJayM, 2023-04-04:
I currently have a problem that if the wifi is not great a request will hang and hold up the whole script. Is there a way to add a timeout to the request? I don't want to use async requests as the program shouldn't keep running without the current data but i would like to retry the requests until a response is received from the server.

Is there a way to do that?

massimosala, 2023-06-28:
Hi

I have rewritten the library, with some improvements and timeouts.

Do you want to test it ?

After your feedback, I will open source it and propose the new version to the Micropython mantainers.

TaylorJayM, 2023-06-29:
> Hi
> 
> I have rewritten the library, with some improvements and timeouts.
> 
> Do you want to test it ?
> 
> After your feedback, I will open source it and propose the new version to the Micropython mantainers.
> 

Sure. Is it your repo?

massimosala, 2023-07-07:
> > Hi
> > I have rewritten the library, with some improvements and timeouts.
> > Do you want to test it ?
> > After your feedback, I will open source it and propose the new version to the Micropython mantainers.
> 
> Sure. Is it your repo?

Hi Taylor
It isn't published on a public repo.

For these first users' feedback, I prefer to "talk" by email.
Write me: massimo.sala.71 AT gmail.com

MartinPacker, 2023-08-19:
+1 on wanting a settable timeout. I suspect an os -2 I'm getting is because of a (short) timeout (hard coded) value.

MartinPacker, 2023-09-15:
So a `try` / `except` block stops the -2 from crashing the script. I could still use a shorter timeout. My failures are generally because the server I'm GETting from and POSTing to are on the wrong network in the house. Should be quick to establish and the current timeout isn't.

prabhu-yu, 2024-11-01:
hi @mstaffa  @MartinPacker @TaylorJayM 
I have submitted a fix here.
https://github.com/micropython/micropython-lib/pull/890/files