micropython/micropython-lib #648: Undefined function in CBOR library

Tangerino, 2023-04-20:
In CBOR library there are a couple of unidentified functions such as:

`unpack_float16`
```python
def decode_float16(decoder):
    payload = decoder.read(2)
    return unpack_float16(payload)
```

and

`undefined`

```python
special_decoders = {
    20: lambda self: False,
    21: lambda self: True,
    22: lambda self: None,
    23: lambda self: undefined,
    24: decode_simple_value,
    25: decode_float16,
    26: decode_float32,
    27: decode_float64,
    31: lambda self: break_marker,
}
```

jimmo, 2023-07-21:
@iabdalkader hi Ibrahim, would you be able to take a look at this? Thanks

iabdalkader, 2023-07-21:
It seems it was not implemented in the original library, I have no idea why, but it should be easy to add.

jonnor, 2024-08-25:
Anyone want to make a MR for this functionality? :)