micropython/micropython-lib #652: traceback.format_exc failed due to sys.exc_info not exist

zcattacz, 2023-05-05:
Is this expected? not design to work on MCU ports ?
```
>>> try:
...     1/0
... except:
...     print(traceback.format_exc())
...     
...     
... 
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "traceback.py", line 27, in format_exc
AttributeError: 'module' object has no attribute 'exc_info'
```

andrewleech, 2023-05-05:
I'm not sure, I haven't tried the traceback module from here it might be out of date.

Take a look at this, it might be what you want: https://docs.micropython.org/en/latest/library/sys.html#sys.print_exception

nkrba, 2024-09-25:
Tried the latest micropython module from this repo, didn't work. Thanks to @andrewleech, the above link worked!
```
try:
  ...
except Exception as exc:
  print_exception(exc)
```