micropython/micropython-lib #668: urequests doesn't handle basic auth formatted URLs correctly

0xVeles, 2023-05-24:
Per [RFC1738](https://www.ietf.org/rfc/rfc1738.txt) you can supply a username and password for basic auth as part of the URL in the format: `http(s)://username:password@example.com` however urequests interprets any colon following the protocol to be delimiting a host and port, [as seen here.](https://github.com/micropython/micropython-lib/blob/7128d423c2e7c0309ac17a1e6ba873b909b24fcc/python-ecosys/urequests/urequests.py#L72)

Obviously it's simple to provide basic auth as a header instead, but it's probably best to be RFC compliant when possible.

massimosala, 2023-06-28:
Probably it isn't implemented because
- passing clear text credentials in the URL is a bad idea
- there are modern authentication options available, this is becoming obsolete.

Just for curiosity:  are there running servers using https://user:pass@... ?

massimosala, 2023-06-28:
Hi

I have rewritten the library, with some improvements and timeouts.

Do you want to test it ?

After your feedback, I will open source it and propose the new version to the Micropython mantainers.

jonnor, 2024-08-25:
I think that parsing username:password out of URLs can be a separate function, which extracts the relevant information and a cleaned URL. And those that the few that need it can copy-paste it into their project.

smithps, 2024-11-18:
I just found this bug report as I am trying to implement DDNS on a PicoW - the URL specified by dyndns is as follows;

`https://{user}:{updater client key}@members.dyndns.org/v3/update?hostname={hostname}&myip={IP Address}`

Which although not using a password as such, is still passing a plaintext "key". 

Out of curiosity and in order to be able to handle such formatted URLs (even if they aren't recommended - it's obvious that they are still in use)....would it be sensible to search backwards from the first '/' looking for a port number and/or if the value found after the ':' is not numeric to ignore it?

smithps, 2024-11-18:
> Probably it isn't implemented because
> 
> * passing clear text credentials in the URL is a bad idea
> * there are modern authentication options available, this is becoming obsolete.
> 
> Just for curiosity: are there running servers using https://user:pass@... ?

Yes; DynDNS (Part of Oracle) use this for updating Dynamic DNS records see their help article [here](https://help.dyn.com/remote-access-api/perform-update/)