micropython/micropython-lib #679: mip: Dependency of copy not installed

sandyscott, 2023-06-13:
When I installed the `copy` package via mip (on attached host PC, not running on the device), the dependecy `types` was not installed.

To Reproduce:

Raspberry Pi Pico running v1.20.0
PC running Python 3.10, pip 3.10, mpremote 1.20.0
```shell
$ mpremote connect com14 mip install copy
Install copy
Installing copy (latest) from https://micropython.org/pi/v2 to /lib
Installing: /lib/copy.mpy
Done
$ mpremote repl
Connected to MicroPython at COM14
Use Ctrl-] or Ctrl-x to exit this shell
>>> import copy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "copy.py", line 51, in <module>
ImportError: no module named 'types'
>>>
```

andrewleech, 2023-06-13:
Thanks for the report, it seems the dependency on types hasn't been declared in the manifest https://github.com/micropython/micropython-lib/blob/master/python-stdlib/copy/manifest.py

These manifests were automatically generated from the earlier setup.py based installer system which was pretty inconsistent in quality, so it's not surprising that some things like this have been missed

andrewleech, 2023-06-14:
Yes I added an extra comment to this PR referring to a proposed fix to the types package I submitted previously in PR #646

closed 2023-07-21 by dpgeorge