micropython/micropython-lib #688: aiorepl:  SyntaxError: invalid syntax on line 29

inovatorius, 2023-06-19:
Hello!

I used the example found here:
https://github.com/micropython/micropython-lib/tree/master/micropython/aiorepl

```
Traceback (most recent call last):
File "main.py", line 2, in <module>
File "aiorepl/aiorepl.py", line 29
SyntaxError: invalid syntax
MicroPython v1.20.0-238-g2a4b356ac-dirty on 2023-06-18; ESP module (1M) with ESP8285
```

jimmo, 2023-06-20:
> ESP module (1M) with ESP8285

The 1M (and smaller) builds for ESP826x don't enable f-strings.

I probably shouldn't have used f-strings in aiorepl and just used str.format which is supported on all boards. Raised https://github.com/micropython/micropython-lib/pull/689 to fix (you can apply the same fix to the copy of aiorepl.py on your board).

inovatorius, 2023-06-20:
Thank you @jimmo!

Are there any benefits enabling and using f-strings in comparison to str.format?

jimmo, 2023-06-22:
> Are there any benefits enabling and using f-strings in comparison to str.format?

@inovatorius I much prefer using f-strings over str.format because it's easier to read/write. But it takes extra code size to support this in the MicroPython firmware, so we don't included it on boards that have smaller flash (like the 1MiB ESP8266). The 2MiB+ ESP8266 firmware does include it.

PR #689 (2023-07-21): aiorepl: Replace f-string with str.format.
f-strings aren't enabled on all builds (e.g. low-flash ESP8266).

Fixes #688

_This work was funded through GitHub Sponsors._
closed 2023-07-21 by dpgeorge