micropython/micropython-lib #700: uaiohttpclient does not support POST/PUT data

GM-Script-Writer-62850, 2023-07-08:
Usage: `await uaiohttpclient.request(method,url)`  - https://github.com/micropython/micropython-lib/blob/7128d423c2e7c0309ac17a1e6ba873b909b24fcc/micropython/uaiohttpclient/uaiohttpclient.py#L65
```
method = "POST"
url = "http://10.0.0.69:8080/input.php"
data = '{"foo":"bar"}'
```

Function has no way to take data

rroemhild, 2023-07-22:
I did the same, then found your issue about it :see_no_evil: 

Here is my attempt, only json, no header support and allow to use a port other than 80.

```python
import uasyncio as asyncio

class ClientResponse:
    def __init__(self, reader):
        self.content = reader

    def read(self, sz=-1):
        return (yield from self.content.read(sz))

    def __repr__(self):
        return "<ClientResponse %d %s>" % (self.status, self.headers)

class ChunkedClientResponse(ClientResponse):
    def __init__(self, reader):
        self.content = reader
        self.chunk_size = 0

    def read(self, sz=4 * 1024 * 1024):
        if self.chunk_size == 0:
            l = yield from self.content.readline()
            # print("chunk line:", l)
            l = l.split(b";", 1)[0]
            self.chunk_size = int(l, 16)
            # print("chunk size:", self.chunk_size)
            if self.chunk_size == 0:
                # End of message
                sep = yield from self.content.read(2)
                assert sep == b"\r\n"
                return b""
        data = yield from self.content.read(min(sz, self.chunk_size))
        self.chunk_size -= len(data)
        if self.chunk_size == 0:
            sep = yield from self.content.read(2)
            assert sep == b"\r\n"
        return data

    def __repr__(self):
        return "<ChunkedClientResponse %d %s>" % (self.status, self.headers)

def request_raw(method, url, json=None):
    data = None

    try:
        proto, dummy, host, path = url.split("/", 3)
    except ValueError:
        proto, dummy, host = url.split("/", 2)
        path = ""
    if proto == "http:":
        port = 80
    else:
        raise ValueError("Unsupported protocol: " + proto)

    if ":" in host:
        host, port = host.split(":", 1)
        port = int(port)

    reader, writer = yield from asyncio.open_connection(host, port)
    # Use protocol 1.0, because 1.1 always allows to use chunked transfer-encoding
    # But explicitly set Connection: close, even though this should be default for 1.0,
    # because some servers misbehave w/o it.
    yield from writer.awrite(b"{} /{} HTTP/1.0\r\nHost: {}\r\nUser-Agent: compat\r\n".format(method, path, host))

    if json is not None:
        import ujson

        data = ujson.dumps(json)
        yield from writer.awrite(b"Content-Type: application/json\r\n")
        yield from writer.awrite(b"Content-Length: {}\r\n".format(len(data)))

    yield from writer.awrite(b"Connection: close\r\n\r\n")

    if data:
        yield from writer.awrite(data)

    # yield from writer.aclose()
    return reader

def request(method, url, json=None):
    redir_cnt = 0
    redir_url = None
    while redir_cnt < 2:
        reader = yield from request_raw(method, url, json)
        headers = []
        sline = yield from reader.readline()
        sline = sline.split(None, 2)
        status = int(sline[1])
        chunked = False
        while True:
            line = yield from reader.readline()
            if not line or line == b"\r\n":
                break
            headers.append(line)
            if line.startswith(b"Transfer-Encoding:"):
                if b"chunked" in line:
                    chunked = True
            elif line.startswith(b"Location:"):
                url = line.rstrip().split(None, 1)[1].decode("latin-1")

        if 301 <= status <= 303:
            redir_cnt += 1
            yield from reader.aclose()
            continue
        break

    if chunked:
        resp = ChunkedClientResponse(reader)
    else:
        resp = ClientResponse(reader)
    resp.status = status
    resp.headers = headers
    return resp
```

Works for me since a few days without problems

jonnor, 2024-08-25:
`uiaiohttpclient` has now been replaced by `aiohttp`. And it now implements POST/PUT, thanks to https://github.com/micropython/micropython-lib/pull/778

So I believe this issue can be closed.

closed 2024-08-25 by mattytrentini