micropython/micropython-lib #701: How to set ATT MTU size in aioble?

rivendell1984, 2023-07-19:
Hello, is there API to set ATT MTU size in aioble?

jimmo, 2023-07-19:
Please see this discussion thread: https://github.com/orgs/micropython/discussions/11945#discussioncomment-6368946

This should be added to the readme.

rivendell1984, 2023-07-19:
> Please see this discussion thread: https://github.com/orgs/micropython/discussions/11945#discussioncomment-6368946
> 
> This should be added to the readme.

Thanks a lot for your response.

The peripheral device has over 100 bytes in characteristics, I can read the value correctly by nRF connect. 

But when I'm using aioble to develop central with 

`aioble_conn.exchange_mtu(new_mtu_length) `

it still read 20 bytes 
The code is below:

```
async def find_device():
    async with aioble.scan(5000, interval_us=12000, window_us=10000, active=True) as scanner:
        async for result in scanner:
            if result.name() == "my-device" and _CUSTOM_SERVICE_UUID in result.services():
                print(result, result.name(), result.rssi, result.services())
                return result.device
    return None

async def keep_finding_device():
    device = await find_device()
    while not device:
        device = await find_device()
    
    ble = bluetooth.BLE()
    print("MTU = ",ble.config('mtu')) #MTU =  256

    try:
        print("Connecting to", device)
        connection = await device.connect()
     except asyncio.TimeoutError:
          print('Timeout during connection')
        
    async with connection:
            try:
                connection.exchange_mtu(128) #It doesn't work
                service = await connection.service(_CUSTOM_SERVICE_UUID)
                characteristic = await service.characteristic(_CUSTOM_CHAR_UUID)
                data = await characteristic.read()
                print("read data", data)
             except asyncio.TimeoutError:
                print("Timeout discovering services/characteristics")
```

rivendell1984, 2023-07-19:
@jimmo Thanks a lot, it works

It hasn't added to readme. the code snippet is below, which works well:

```
  try:
    print("Connecting to", device)
    connection = await device.connect()
  except asyncio.TimeoutError:
    print('Timeout during connection')
        
  async with connection:
    try:
      await connection.exchange_mtu(128)
      service = await connection.service(_CUSTOM_SERVICE_UUID)
...
```

closed 2023-07-20 by rivendell1984