micropython/micropython-lib #717: where the 'uio' come from?

yingshaoxo, 2023-08-17:
When I look at the io module, it imported a module called 'uio'

But I did not found it anywhere, where that thing comes from?

mattytrentini, 2023-08-17:
Until [very recently](https://github.com/orgs/micropython/discussions/12078), MicroPython used a 'u' prefix on a number of libraries to indicate that it was a subset - and sometimes _different_ - to the CPython library of the same name. 

Nightly builds, and the impending v1.21 release, will drop the 'u' prefix but provide backwards compatibility (ie `import uio` will continue to work though `import io` should be preferred).

jonnor, 2024-08-25:
This question looks to be answered. Proposing to close this issue.

closed 2024-08-25 by mattytrentini