micropython/micropython-lib #732: 'module' object has no attribute 'EPIPE'

LUCKandII, 2023-09-20:
Hi, I have questions about using this library `errno`,

* I can import this library `errno`,but how do I know how the error flags correspond to the numbers? I refer to the source code here: [python-stdlib/errno/errno.py](https://github.com/micropython/micropython-lib/blob/e6b89eafa3b86d2e8e405450377d459600a30cd6/python-stdlib/errno/errno.py) but found a problem:

```
>>> import errno
>>> errno.ETIMEDOUT
116
>>> errno.EINPROGRESS
119
>>> errno.EPIPE
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'EPIPE'
```

* But in errno.py, `errno.ETIMEDOUT` should be `110`,  `errno.EINPROGRESS` should be `115`, `errno.EPIPE` should be `32`. Did I refer to the wrong source code?

This problem will affect the porting and development of some libraries especially using `errno.EPIPE` and I am confuse. I 'd love to know where the exact source code is. 

* hardware: ESP32 / WROOM
* micropython version: [ESP32_GENERIC-20230426-v1.20.0.bin](https://micropython.org/resources/firmware/ESP32_GENERIC-20230426-v1.20.0.bin)

andrewleech, 2023-09-22:
Again, there's a builtin version of errno as well as the micropython-lib copy: https://github.com/micropython/micropython/blob/a3862e726728725248d42ea9173675d523c13911/py/moderrno.c#L38

LUCKandII, 2023-09-23:
@andrewleech Thanks! I guess I understand:
* In this folder `micropython\ports\esp32`, there is no such a file named `moderrno.c` nor `MICROPY_PY_ERRNO_LIST` defined in `micropython\ports\esp32\mpconfigport.h`, so we should refer to `micropython\py\moderrno.c` and the `MICROPY_PY_ERRNO_LIST` in it. So the `errno.EPIPE` code is NOT included.
* As for ERRNO numbers, referring to the file `micropython\ports\esp32\mpconfigport.h`, [here](https://github.com/micropython/micropython/blob/a3862e726728725248d42ea9173675d523c13911/ports/esp32/mpconfigport.h#L54C8-L54C8) has explained errno.h source code for ESP32. To find the answer, just download `ESP-IDF` and look into `xtensa-esp32-elf/sys-include/sys`
* As for micropython feature and libraries, sitll referring to the file `micropython\ports\esp32\mpconfigport.h`, it clearly illustrates which libraries are from `port` and from `extmod`

closed 2023-09-24 by LUCKandII