micropython/micropython-lib #733: Where is

LUCKandII, 2023-09-20:


andrewleech, 2023-09-20:
Most builtin libraries are implemented in C in the main micropython repo, eg. https://github.com/micropython/micropython/blob/master/extmod/modtime.c

FWIW the `u` prefix has recently been dropped by default, so while `utime` still works it's intended to just use `time` now!

LUCKandII, 2023-09-21:
> Most builtin libraries are implemented in C in the main micropython repo, eg. https://github.com/micropython/micropython/blob/master/extmod/modtime.c
> 
> FWIW the `u` prefix has recently been dropped by default, so while `utime` still works it's intended to just use `time` now!

@andrewleech Thanks! Am I understanding this correctly:

* `import time`, the source code is in [micropython/blob/extmod/modtime.c](https://github.com/micropython/micropython/blob/a3862e726728725248d42ea9173675d523c13911/extmod/modtime.c), When I use this `time.localtime()` in microython, `mp_obj_t time_localtime(size_t n_args, const mp_obj_t *args)` is called in `extmod/modtime.c`. 
* Some methods for different platforms is in `micropython/ports`. When I use `from machine import RTC` on ESP32 board, `micropython/ports/esp32/machine_rtc.c` is exactly used.
* Some Cross-platform librarys is in [micropython-lib](https://github.com/micropython/micropython-lib) repo, such as `import ntptime`, `micropython-lib/micropython/net/ntptime/ntptime.py` is exactly used.

But I still have a question: What's the relationship between `micropython/blob/extmod/machine_i2c.c` and `micropython/ports/esp32/machine_i2c.c` these? When I `from machine import i2c` on ESP32 board, which file is exactly used?

andrewleech, 2023-09-21:
Yep you've got it right there. Things in micropython-lib can be mip installed by users or frozen into a build, some boards/ports have a bunch of libraries specified in their manifest.py.

The question about port vs extmod versions of modules is a tricky one...

There's a goal to get all ports as consistent as possible with built-in drivers in the machine module; common versions of these are generally written in the extmod dir that can be used on any port.

However... many / most ports have some device specific drivers in machine that have features not yet available in the (newer) extmod versions so still have their own copy in the port folder. Really the only way to tell which one is used is to look at the Makefile/cmakelist files for the port to see what's included.

It's a work in progress to gradually consolidate the machine drivers so they can cover enough port driven features to eventually remove port specific copies.

LUCKandII, 2023-09-22:
@andrewleech __Thank you! It really helps me a lot!__
Here is a related problem using library functions that I found that the micropython output from the command line does not match the source code in #732 .Would you mind to have a look?😊

closed 2023-09-23 by LUCKandII