micropython/micropython-lib #734: aioble: inconsistent service registering process with pi pico w

griffinsteffy19, 2023-09-24:
I am running a bluetooth service that currently just as a rx service with write and capture both set to true. I am getting in consent results when reloading the device after awhile without being powered. I continue to get the following error...
`
  File "App/TX1/bleService.py", line 50, in __init__
  File "aioble/server.py", line 328, in register_services
  File "aioble/core.py", line 38, in ensure_active
OSError: [Errno 110] ETIMEDOUT
`

```python
# Ble Service Module

# Module Includes

# Common Application Modules
from App.common.logModule import Log

# Core Libaries
import bluetooth
import uasyncio as asyncio
from micropython import const
import aioble
import random
import struct

# Ble Provision Service Logger
log = Log("bleS", 'TRACE')
# 8c721902-c5b2-4dab-b935-3e512a679277
# 294ef9b0-e3cb-4f98-8f5b-33902814bcf8
# c06f9b3f-d897-40b4-9fb2-2c3ec66ceacc

# Module Defines
RX_TIMEOUT_MS = 5000

# Uart Service UUID
UART_SERVICE_UUID = bluetooth.UUID('8c721902-c5b2-4dab-b935-3e512a679277')

# RX Characteristic UUID
UART_RX_CHARACTERISTIC_UUID = bluetooth.UUID(
    '294ef9b0-e3cb-4f98-8f5b-33902814bcf8')

# How frequently to send advertising beacons.
_ADV_INTERVAL_MS = 250_000

class BLEService:
    def __init__(self, dataCb, advName="CTdvc"):
        self.name = advName
        self.dataCb = dataCb
        self.isStarted = False
        self.advTask = None
        self.rxCbTask = None

        # Register GATT server.
        self.uart_service = aioble.Service(UART_SERVICE_UUID)
        self.rx_characteristic = aioble.Characteristic(
            self.uart_service, UART_RX_CHARACTERISTIC_UUID, write=True, capture=True
        )
        log.debug('Registering Services...')
        aioble.register_services(self.uart_service)
        log.info('Services Registered')

    async def rxCb(self):
        while True:
            await self.rx_characteristic.written()
            try:
                data = self.rx_characteristic.read().decode('utf-8')
                print(f"written: {data}")
                if(None != self.dataCb):
                    await self.dataCb(data)
            except UnicodeError:
                log.error('Unicode Error during decode')
        self.rxCbTask = None

    async def advertisingTask(self):
        while True:
            async with await aioble.advertise(
                _ADV_INTERVAL_MS,
                name=self.name,
                services=[UART_SERVICE_UUID],

            ) as connection:
                print("Connection from", connection.device)

                await connection.disconnected()

        self.advTask = None

    def started(self):
        return self.isStarted

    async def start(self):
        log.trace('start')
        if(not self.isStarted):

            log.debug("staring BLE with name=%s" % self.name)
            self.rxCbTask = asyncio.create_task(self.rxCb())
            self.advTask = asyncio.create_task(self.advertisingTask())
            self.isStarted = True
        else:
            log.debug('already started')

        while True:
            asyncio.sleep(0.5)

    async def stop(self):
        log.trace('stop')
        self.isStarted = False

        if(None != self.advTask):
            self.advTask.cancel()
            self.advTask = None

        if(None != self.rxCbTask):
            self.rxCbTask.cancel()
            self.rxCbTask = None

```

Above this module I have initialize it, then call start within a running while loop task

griffinsteffy19, 2023-09-24:
It looks like this might be coming form the underlying `bluetooth` library:

```bash
>>> import bluetooth
>>> ble = bluetooth.BLE()
>>> ble.active(True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
```

Here is the UF2 I am running: [v1.20.0-198-g0eacdeb1c](https://datasheets.raspberrypi.com/soft/micropython-firmware-pico-w-130623.uf2?_gl=1*5zvp68*_ga*MTY0MjUwOTIyNC4xNjk1NTIwMzgz*_ga_22FD70LWDS*MTY5NTU4MTUzMC42LjAuMTY5NTU4MTUzMC4wLjAuMA..)

MattyBoy4444, 2023-10-25:
Did you figure out the problem?

griffinsteffy19, 2023-10-25:
No, most consistent way to fix the issue is to reload the uf2 file.

griffinsteffy19, 2023-10-26:
I haven't ran any of the recent preview builds (after 1.21.0). I'll do some testing tomorrow with the latest preview and see what happens.

griffinsteffy19, 2024-11-14:
Leaving a final update 1 Year Later. It seems like that fix mentioned has solved it, but don't have any A/B tests to prove it. I'll close this issue for now.

closed 2024-11-14 by griffinsteffy19