micropython/micropython-lib #741: mem leak in urequests.post()

gamename, 2023-09-30:
Hi,

After about 6 POSTs to an HTTPS url, my Pico will crash.  

Just to be certain, I loaded the latest `urequests` version. Same result.  

Here is some output:
```python
MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import mip
>>> mip.install('urequests')
>>>
MPY: soft reboot
WIFI: Attempting network connection
WIFI: Successfully connected to network
MAIN: Starting event loop
MEM: Count: 1 Free memory: 180224
MEM: Count: 2 Free memory: 142160
MEM: Count: 3 Free memory: 104096
MEM: Count: 4 Free memory: 66032
MEM: Count: 5 Free memory: 27968
MEM: Count: 6 Free memory: 17056
Traceback (most recent call last):
  File "main.py", line 58, in <module>
  File "main.py", line 52, in main
  File "urequests.py", line 184, in post
  File "urequests.py", line 93, in request
OSError: [Errno 12] ENOMEM
MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>>
```

Here is the code to reproduce the error:
```python
import gc
import time

import network
import ntptime
import urequests as requests
import utime
from machine import Pin, reset

import secrets

REQUEST_HEADER = {'content-type': 'application/json'}

def pico_wifi_connect(wlan, ssid, password, connection_attempts=10, sleep_seconds_interval=3):
    led = Pin("LED", Pin.OUT)
    led.off()
    print("WIFI: Attempting network connection")
    wlan.active(True)
    time.sleep(sleep_seconds_interval)
    counter = 1
    wlan.connect(ssid, password)
    while not wlan.isconnected():
        print(f'WIFI: Attempt {counter} of {connection_attempts}')
        time.sleep(sleep_seconds_interval)
        counter += 1
        if counter > connection_attempts:
            print("WIFI: Max connection attempts exceeded. Resetting microcontroller")
            time.sleep(1)  # Gives the system time enough to print above msg to screen
            reset()
    led.on()
    print("WIFI: Successfully connected to network")

def main():
    gc.enable()
    network.hostname('mem_leak_test')
    #
    ap_if = network.WLAN(network.AP_IF)
    ap_if.active(False)
    #
    wlan = network.WLAN(network.STA_IF)
    pico_wifi_connect(wlan, secrets.SSID, secrets.PASSWORD)
    #
    # Sync system time with NTP
    ntptime.settime()
    #
    print("MAIN: Starting event loop")
    counter = 1
    while True:
        print(f"MEM: Count: {counter} Free memory: {gc.mem_free()}")
        requests.post(secrets.REST_API_URL + 'closed', headers=REQUEST_HEADER)
        counter += 1
        utime.sleep(30)

if __name__ == "__main__":
    main()
```

jimmo, 2023-10-04:
@gamename You need to close the response object.

```
resp = requests.post(secrets.REST_API_URL + 'closed', headers=REQUEST_HEADER)
resp.close()
```

This gets done for you if you access `resp.content` or `resp.json()` too.

closed 2023-10-04 by jimmo

gamename, 2023-10-04:
@jimmo Thanks!