micropython/micropython-lib #742: the json library in python-stdlib requires some constants in re module which is not available in the embeded re module of micropython

versaloon, 2023-10-01:
`
/ # mpy
MicroPython v1.20.0-348-g24a6e951e-dirty on 2023-08-13; linux [] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import mip
>>> mip.install('json')
Installing json (latest) from https://micropython.org/pi/v2 to /usr/micropython/lib
Copying: /usr/micropython/lib/json/__init__.mpy
Copying: /usr/micropython/lib/json/encoder.mpy
Copying: /usr/micropython/lib/json/scanner.mpy
Copying: /usr/micropython/lib/json/tool.mpy
Copying: /usr/micropython/lib/json/decoder.mpy
Done
>>> import json
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "json/__init__.py", line 114, in <module>
  File "json/decoder.py", line 6, in <module>
  File "json/scanner.py", line 13, in <module>
AttributeError: 'module' object has no attribute 'VERBOSE
`

andrewleech, 2023-10-01:
This micropython-lib json library is quite old, I'm guessing it was written against the ffi re module (unix only version).

Are there features in this json that the built-in one is missing?

versaloon, 2023-10-01:
> This micropython-lib json library is quite old, I'm guessing it was written against the ffi re module (unix only version).
> 
> Are there features in this json that the built-in one is missing?

the json library depends on re library, and need constants below, which is not in embedded re module implementation:
re.VERBOSE
re.MULTILINE
re.DOTALL

andrewleech, 2023-10-02:
I meant are you trying to use the mip installed `json` to do something that doesn't work on the builtin `import json`?

versaloon, 2023-10-07:
No, I just do a test.
I have switched to embedded json.

closed 2023-10-07 by versaloon