micropython/micropython-lib #743: Problem with installing micropython-machine?

somniumLux, 2023-10-01:
I have run a program before on my RPI Pico Zero W with the machine module in PyCharm but now I cannot set it up again. When running the program, I get an error: "ModuleNotFoundError: No module named 'machine'"

Running both `pip install micropython-machine` and `pip install machine` comes with errors. (Although I don't know what is the difference between those two. I am new in Python.)

Below are outputs in terminal for those two commands and I don't know how to resolve them.

**For `pip install micropython-machine` I get:**

Collecting micropython-machine
  Using cached micropython-machine-0.2.tar.gz (1.7 kB)
ERROR: micropython-machine from https://files.pythonhosted.org/packages/e1/31/a3282942b3e96881393db406130735913c647f0fc9e07a0d60d7b546b06b/micropython-machine-0.2.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.

**For `pip install machine` I get:**

Collecting machine
  Using cached machine-0.0.1-py2.py3-none-any.whl (4.2 kB)
Collecting pycrypto (from machine)
  Using cached pycrypto-2.6.1.tar.gz (446 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: pycrypto
  Building wheel for pycrypto (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [28 lines of output]
      warning: GMP or MPIR library not found; Not building Crypto.PublicKey._fastmath.
      winrand.c
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(31): error C2061: syntax error: identifier 'intmax_t'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(32): error C2061: syntax error: identifier 'rem'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(32): error C2059: syntax error: ';'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(33): error C2059: syntax error: '}'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(35): error C2061: syntax error: identifier 'imaxdiv_t'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(35): error C2059: syntax error: ';'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(45): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(46): error C2146: syntax error: missing ')' before identifier '_Number'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(46): error C2061: syntax error: identifier '_Number'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(46): error C2059: syntax error: ';'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(47): error C2059: syntax error: ')'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(50): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(51): error C2146: syntax error: missing ')' before identifier '_Numerator'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(51): error C2061: syntax error: identifier '_Numerator'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(51): error C2059: syntax error: ';'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(51): error C2059: syntax error: ','
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(53): error C2059: syntax error: ')'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(55): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(61): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(68): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(74): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(81): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(87): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(94): error C2143: syntax error: missing '{' before '__cdecl'
      C:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt\inttypes.h(100): error C2143: syntax error: missing '{' before '__cdecl'
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.37.32822\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycrypto
  Running setup.py clean for pycrypto
Failed to build pycrypto
ERROR: Could not build wheels for pycrypto, which is required to install pyproject.toml-based projects

somniumLux, 2024-01-06:
Hi, somehow it is working now. I reinstalled PyCharm after a while and it succesfully pulls the module. I don't know how or why but compilation works.

Btw, my is Raspberry Pi Pico W, sorry for my mistake.

Thanks for you response :)

ilario, 2024-01-11:
I just observed the same issue on a A64-Olinuxino, running Linux with Python3:

```
pip install micropython-machine
Collecting micropython-machine
  Downloading micropython-machine-0.2.tar.gz (1.7 kB)
ERROR: micropython-machine from https://files.pythonhosted.org/packages/e1/31/a3282942b3e96881393db406130735913c647f0fc9e07a0d60d7b546b06b/micropython-machine-0.2.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
```

If `micropython-machine` is not supposed to be installed via pip, please consider removing it from PyPI https://pypi.org/project/micropython-machine/

andrewleech, 2024-01-11:
What are you trying to achieve here? 

micropython code that uses `machine` is code that's directly controlling hardware - this could never "just work" on desktop python, regardless of which packages are installed.

Though yeah, it probably would be best for all legacy micropython packages on pypi to just be removed, they're not intended to be used at all now that `mip` is the standard package manager.

ilario, 2024-01-12:
> What are you trying to achieve here?
> 
> micropython code that uses `machine` is code that's directly controlling hardware - this could never "just work" on desktop python, regardless of which packages are installed.

Yep, don't worry. I tried to install it before realizing this. 

If you are curious about how I ended up trying to install it: What happened is that A64-Olinuxino is not recognized by [Adafruit_Python_PlatformDetect](https://github.com/adafruit/Adafruit_Python_PlatformDetect), so that when trying to use Adafruit Blinka's busio for I2c, there are various imports until getting to this switch:
https://github.com/adafruit/Adafruit_Blinka/blob/60fa43167c45c1ab162292dcf26d6dce9e001851/src/busio.py#L139-L149
As the A64-Olinuxino is not known by Adafruit_Python_PlatformDetect, the code uses micropython things. I suppose that the meaningful solution would be that anyone from @OLIMEX or any member of their community add the support of A64-Olinuxino in PlatformDetect.

> Though yeah, it probably would be best for all legacy micropython packages on pypi to just be removed, they're not intended to be used at all now that `mip` is the standard package manager.

Yep, for me it was confusing to have micropython-machine there but to not be able to install it with the usual tools.

jonnor, 2024-08-25:
The problems here seem to have been resolved/clarified. Proposing to close this issue

closed 2024-08-25 by mattytrentini

ilario, 2024-08-26:
> The problems here seem to have been resolved/clarified. Proposing to close this issue

Clarified it is, thanks.
But for avoiding more users to be confused, I would suggest to remove the package from the PyPI website https://pypi.org/project/micropython-machine/